llvm/circt #1512: [FIRRTL] (core dump) ShiftAmt <= BitWidth && "Invalid shift amount"

drom, 2021-08-03:
The following FIRRTL program:

```python
; seed: 153
circuit top_mod :
  module mod_0 :
    input clock: Clock
    input arst: AsyncReset
    input inp_cb: UInt<67>
    output fsm_97_next: UInt<4>
    reg fsm_97_state: UInt, clock with: (reset => (arst, UInt(0)))
    node tmp24 = xor(inp_cb, asUInt(SInt<14>(42)))
    node tmp73 = dshl(UInt<20>(42), head(tmp24, 5))
    when eq(fsm_97_state, UInt<4>(10)) :
      when xorr(tmp73) :
        fsm_97_next <= UInt<4>(0)
      else :
        fsm_97_next <= fsm_97_state
    else :
      fsm_97_next <= fsm_97_state
    fsm_97_state <= fsm_97_next
  module top_mod :
    inst U0 of mod_0
    U0 is invalid
```

Compiled with `valgrind -q $FIRTOOL $INPUT --lower-to-hw --infer-widths --mlir-timing --verilog -o=$VFILE2` fails with the following core dump:

```cpp
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/APInt.h:995: void llvm::APInt::lshrInPlace(unsigned int): Assertion `ShiftAmt <= BitWidth && "Invalid shift amount"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --infer-widths --mlir-timing --verilog -o=a_top_mod.fir.mfc.v
 #0 0x00000000004e2be7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004e2caa PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004e0bf1 llvm::sys::RunSignalHandlers() (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:97:20
 #3 0x00000000004e2612 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x000000000488b970 __restore_rt (/lib64/libpthread.so.0+0x13970)
 #5 0x0000000004a93322 raise (/lib64/libc.so.6+0x3d322)
 #6 0x0000000004a7c864 abort (/lib64/libc.so.6+0x26864)
 #7 0x0000000004a7c749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x0000000004a8b9d6 (/lib64/libc.so.6+0x359d6)
 #9 0x00000000004fc5c3 llvm::APInt::lshrInPlace(unsigned int) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/APInt.h:996:21
#10 0x000000000050e77b llvm::APInt::lshr(unsigned int) const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/APInt.h:990:12
#11 0x00000000008d9e3a circt::firrtl::BitsPrimOp::fold(llvm::ArrayRef<mlir::Attribute>) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/FIRRTLFolds.cpp:973:63
#12 0x00000000008b5dbb mlir::LogicalResult mlir::Op<circt::firrtl::BitsPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::InferTypeOpInterface::Trait, mlir::MemoryEffectOpInterface::Trait>::foldSingleResultHook<circt::firrtl::BitsPrimOp>(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:1723:53
#13 0x00000000008aef34 _ZZN4mlir2OpIN5circt6firrtl10BitsPrimOpEJNS_7OpTrait10ZeroRegionENS4_9OneResultENS4_14OneTypedResultINS2_8UIntTypeEE4ImplENS4_13ZeroSuccessorENS4_10OneOperandENS_20InferTypeOpInterface5TraitENS_23MemoryEffectOpInterface5TraitEEE17getFoldHookFnImplIS3_EENSt9enable_ifIXaasrN4llvm11disjunctionIJSt7is_sameINS6_IT_EENS5_ISN_EEESM_ISO_SO_ESM_ISO_NS9_4ImplISN_EEESM_ISO_NSB_ISN_EEESM_ISO_NSC_ISN_EEESM_ISO_NSD_5TraitISN_EEESM_ISO_NSF_5TraitISN_EEEEEE5valuesrNSK_6detail8detectorIvNSH_22has_single_result_foldEJSN_EE7value_tE5valueENSK_15unique_functionIKFNS_13LogicalResultEPNS_9OperationENSK_8ArrayRefINS_9AttributeEEERNSK_15SmallVectorImplINS_12OpFoldResultEEEEEEE4typeEvENKUlS1E_S1H_S1L_E_clES1E_S1H_S1L_ /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:1685:12
#14 0x00000000008c7fed mlir::LogicalResult llvm::detail::UniqueFunctionBase<mlir::LogicalResult, mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&>::CallImpl<std::enable_if<(llvm::disjunction<std::is_same<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp>, mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::OpTrait::ZeroSuccessor<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::OpTrait::OneOperand<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::InferTypeOpInterface::Trait<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > >, circt::firrtl::BitsPrimOp<mlir::OpTrait::ZeroRegion<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >, mlir::MemoryEffectOpInterface::Trait<mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> > > >::value) && (llvm::detail::detector<void, mlir::Op<circt::firrtl::BitsPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::InferTypeOpInterface::Trait, mlir::MemoryEffectOpInterface::Trait>::has_single_result_fold, mlir::OpTrait::OneResult<circt::firrtl::BitsPrimOp> >::value_t::value), llvm::unique_function<mlir::LogicalResult (mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const> >::type mlir::Op<circt::firrtl::BitsPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::InferTypeOpInterface::Trait, mlir::MemoryEffectOpInterface::Trait>::getFoldHookFnImpl<circt::firrtl::BitsPrimOp>()::'lambda'(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const>(void*, mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:217:3
#15 0x000000000080c596 llvm::unique_function<mlir::LogicalResult (mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const>::operator()(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/FunctionExtras.h:405:3
#16 0x000000000080ba3e mlir::AbstractOperation::foldHook(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:131:3
#17 0x00000000008045b6 mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:556:30
#18 0x000000000064c4c1 (anonymous namespace)::IMConstPropPass::visitOperation(mlir::Operation*) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:608:13
#19 0x000000000064b093 (anonymous namespace)::IMConstPropPass::runOnOperation() /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:304:48
#20 0x0000000000ba989d mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (.localalias) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:386:27
#21 0x0000000000ba9b13 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (.localalias) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:445:15
[... cut ...]
#32 0x0000000000406c47 executeFirtool(mlir::MLIRContext&) /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:377:30
#33 0x0000000000406e29 main /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:413:31
#34 0x0000000004a7db35 __libc_start_main (/lib64/libc.so.6+0x27b35)
#35 0x000000000040584e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 43: 20930 Aborted                 (core dumped) valgrind -q $FIRTOOL $INPUT --lower-to-hw --infer-widths --mlir-timing --verilog -o=$VFILE2
```

seldridge, 2021-08-03:
Looks like a dupe of #1512. There's an invalid getting passed into an `xor`. This should be fixed with ee0f736e52f01e964d4f55de66b9197a244155a2.  On `main`, I'm now getting (`firtool -verilog`):

```verilog
module mod_0(
  input         clock, arst,
  input  [66:0] inp_cb,
  output [3:0]  fsm_97_next);

  assign fsm_97_next = 4'h0;	// 1512.fir:2:10, :16:19
endmodule

module top_mod();
  wire [3:0] U0_fsm_97_next;	// 1512.fir:19:5

  mod_0 U0 (	// 1512.fir:19:5
    .clock       (1'h0),	// 1512.fir:20:5
    .arst        (1'h0),	// 1512.fir:20:5
    .inp_cb      (67'h0),	// 1512.fir:20:5
    .fsm_97_next (U0_fsm_97_next)
  );
endmodule
```

closed 2021-08-03 by seldridge

lattner, 2021-08-03:
nice