llvm/circt #1513: [FIRRTL] (core dump) Ptr >= BufStart && Ptr <= Buffer->getBufferEnd()

drom, 2021-08-03:
Running `firtool a_top_mod.fir --lower-to-hw --infer-widths --mlir-timing --verilog -o=a_top_mod.fir.mfc.v`

I am getting this error at random:
```
firtool
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/SourceMgr.cpp:97: unsigned int llvm::SourceMgr::SrcBuffer::getLineNumberSpecialized(const char*) const [with T = short unsigned int]: Assertion `Ptr >= BufStart && Ptr <= Buffer->getBufferEnd()' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --infer-widths --mlir-timing --verilog -o=a_top_mod.fir.mfc.v
1.      In-Flight Diagnostics:
    loc("a_top_mod.fir":419:14): error: expected statement to be on its own line
 #0 0x00000000004e2be7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004e2caa PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004e0bf1 llvm::sys::RunSignalHandlers() (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:97:20
 #3 0x00000000004e2612 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x000000000488b970 __restore_rt (/lib64/libpthread.so.0+0x13970)
 #5 0x0000000004a93322 raise (/lib64/libc.so.6+0x3d322)
 #6 0x0000000004a7c864 abort (/lib64/libc.so.6+0x26864)
 #7 0x0000000004a7c749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x0000000004a8b9d6 (/lib64/libc.so.6+0x359d6)
 #9 0x000000000045ac32 unsigned int llvm::SourceMgr::SrcBuffer::getLineNumberSpecialized<unsigned short>(char const*) const /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/SourceMgr.cpp:98:13
#10 0x0000000000457a5c llvm::SourceMgr::SrcBuffer::getLineNumber(char const*) const (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/SourceMgr.cpp:115:50
#11 0x0000000000457d83 llvm::SourceMgr::getLineAndColumn(llvm::SMLoc, unsigned int) const (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/SourceMgr.cpp:188:41
#12 0x00000000005bd4d9 circt::firrtl::FIRLexer::translateLocation(llvm::SMLoc) (.localalias) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRLexer.cpp:172:50
#13 0x0000000000565e25 (anonymous namespace)::FIRParser::translateLocation(llvm::SMLoc) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:189:3
#14 0x000000000056af90 (anonymous namespace)::LazyLocationListener::endStatement((anonymous namespace)::FIRParser&) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:1412:33
#15 0x00000000005719b2 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2124:10
#16 0x000000000057192a (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2115:24
#17 0x0000000000574209 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int)::'lambda'(mlir::Block&)::operator()(mlir::Block&) const /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2545:42
#18 0x000000000057446f (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2549:17
#19 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#20 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#21 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#22 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#23 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#24 0x000000000057192a (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2115:24
#25 0x0000000000574209 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int)::'lambda'(mlir::Block&)::operator()(mlir::Block&) const /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2545:42
#26 0x000000000057446f (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2549:17
#27 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#28 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#29 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#30 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#31 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#32 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#33 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#34 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#35 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#36 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#37 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#38 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#39 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#40 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#41 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#42 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#43 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
#44 0x00000000005745f5 (anonymous namespace)::FIRStmtParser::parseWhen(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2574:37
#45 0x0000000000571ae9 (anonymous namespace)::FIRStmtParser::parseSimpleStmtImpl(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2168:32
#46 0x0000000000571998 (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:2122:36
[... cut ...]
#66 0x0000000000406c47 executeFirtool(mlir::MLIRContext&) /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:377:30
#67 0x0000000000406e29 main /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:413:31
#68 0x0000000004a7db35 __libc_start_main (/lib64/libc.so.6+0x27b35)
#69 0x000000000040584e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 43: 22168 Aborted                 (core dumped) valgrind -q $FIRTOOL $INPUT --lower-to-hw --infer-widths --mlir-timing --verilog -o=$VFILE2
```

seldridge, 2021-08-03:
This stack trace looks very similar to an issue that @teqdruid  reported upstream: https://bugs.llvm.org/show_bug.cgi?id=49568.

lattner, 2021-08-03:
@drom does this bug disappear when you pass --mlir-disable-threading ?

lattner, 2021-08-03:
I can't reproduce this.  @drom can you see if https://github.com/llvm/circt/pull/1515 fixes the issue?

drom, 2021-08-03:
I can't reproduce this issue reliably. It appears at random.
I will try to find conditions, that increase probability of it happen.

lattner, 2021-08-03:
Right, plz apply the PR i linked to. does the random issue still repro for you with that patch?

closed 2021-08-03 by lattner