llvm/circt #1518: [FIRRTL] (false negative) don't touch invalid reg

drom, 2021-08-03:
The following FIRRTL program

```python
; seed: 32
circuit top_mod :
  module top_mod :
    input clock: Clock
    input arst: AsyncReset
    reg tmp107: UInt, clock with: (reset => (arst, UInt(0)))
    tmp107 is invalid
```

and annotation file:

```json
[{"class":"firrtl.transforms.DontTouchAnnotation","target":"~top_mod|top_mod>tmp107"}]
```

Compiled with `firtool` produces this error:

```cpp
a_top_mod.fir:7:12: error: 'firrtl.connect' op LowerToHW couldn't handle this operation
    tmp107 is invalid
           ^
a_top_mod.fir:7:12: note: see current operation: "firrtl.connect"(%6, %3) : (!firrtl.uint<1>, !firrtl.uint) -> ()
a_top_mod.fir:6:52: error: value has an unsupported verilog type '!firrtl.uint<1>'
    reg tmp107: UInt, clock with: (reset => (arst, UInt(0)))
                                                   ^
a_top_mod.fir:6:52: note: see current operation: %0 = "firrtl.constant"() {value = 0 : ui1} : () -> !firrtl.uint<1>
a_top_mod.fir:7:12: error: value has an unsupported verilog type '!firrtl.uint'
    tmp107 is invalid
           ^
a_top_mod.fir:7:12: note: see current operation: %1 = "firrtl.invalidvalue"() : () -> !firrtl.uint
a_top_mod.fir:3:10: error: value has an unsupported verilog type '!firrtl.clock'
  module top_mod :
         ^
a_top_mod.fir:3:10: note: see current operation: %2 = "firrtl.stdIntCast"(%arg0) : (i1) -> !firrtl.clock
a_top_mod.fir:3:10: error: value has an unsupported verilog type '!firrtl.asyncreset'
  module top_mod :
         ^
a_top_mod.fir:3:10: note: see current operation: %3 = "firrtl.stdIntCast"(%arg1) : (i1) -> !firrtl.asyncreset
a_top_mod.fir:6:5: error: value has an unsupported verilog type '!firrtl.uint<1>'
    reg tmp107: UInt, clock with: (reset => (arst, UInt(0)))
    ^
a_top_mod.fir:6:5: note: see current operation: %5 = "firrtl.regreset"(%2, %3, %0) {annotations = [{class = "firrtl.transforms.DontTouchAnnotation"}], name = "tmp107"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<1>) -> !firrtl.uint<1>
a_top_mod.fir:6:52: error: 'firrtl.constant' op cannot emit this operation to Verilog
    reg tmp107: UInt, clock with: (reset => (arst, UInt(0)))
                                                   ^
a_top_mod.fir:6:52: note: see current operation: %0 = "firrtl.constant"() {value = 0 : ui1} : () -> !firrtl.uint<1>
a_top_mod.fir:7:12: error: 'firrtl.invalidvalue' op cannot emit this operation to Verilog
    tmp107 is invalid
           ^
a_top_mod.fir:7:12: note: see current operation: %1 = "firrtl.invalidvalue"() : () -> !firrtl.uint
a_top_mod.fir:3:10: error: 'firrtl.stdIntCast' op cannot emit this operation to Verilog
  module top_mod :
         ^
a_top_mod.fir:3:10: note: see current operation: %2 = "firrtl.stdIntCast"(%arg0) : (i1) -> !firrtl.clock
a_top_mod.fir:3:10: error: 'firrtl.stdIntCast' op cannot emit this operation to Verilog
  module top_mod :
         ^
a_top_mod.fir:3:10: note: see current operation: %3 = "firrtl.stdIntCast"(%arg1) : (i1) -> !firrtl.asyncreset
a_top_mod.fir:6:5: error: 'firrtl.regreset' op cannot emit this operation to Verilog
    reg tmp107: UInt, clock with: (reset => (arst, UInt(0)))
    ^
a_top_mod.fir:6:5: note: see current operation: %5 = "firrtl.regreset"(%2, %3, %0) {annotations = [{class = "firrtl.transforms.DontTouchAnnotation"}], name = "tmp107"} : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<1>) -> !firrtl.uint<1>
a_top_mod.fir:7:12: error: 'firrtl.connect' op cannot emit this operation to Verilog
    tmp107 is invalid
           ^
a_top_mod.fir:7:12: note: see current operation: "firrtl.connect"(%5, %1) : (!firrtl.uint<1>, !firrtl.uint) -> ()
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input   clock,
  input   arst
);
endmodule
```

seldridge, 2021-08-03:
I think you need to compile the SFC version with `firrtl -i Foo.fir --annotation-file Foo.anno.json`. With that, the register should hang around in the SFC to produce the following (with register randomization trimmed using `verilator -E -P`):

```verilog
module top_mod(
  input   clock,
  input   arst
);
  reg  tmp107;
  always @(posedge clock or posedge arst) begin
    if (arst) begin
      tmp107 <= 1'h0;
    end else begin
      tmp107 <= 1'h0;
    end
  end
initial begin
end  
endmodule
```

drom, 2021-08-04:
agree, about SFC. But MFC is the one generating an error.

seldridge, 2021-08-04:
Agreed. I just want to make it clear that there is one issue here (CIRCT is generating an error message) and not a second, separate SFC issue where the SFC is deleting a register it should not be deleting.

seldridge, 2022-07-03:
I tested this on `main` and it looks good now---no crash and the CIRCT output matches SFC.

MFC:
```verilog
  always @(posedge clock or posedge arst) begin
    if (arst)
      tmp107 <= 1'h0;
    else
      tmp107 <= 1'h0;
  end // always @(posedge, posedge)
```

SFC:
```verilog
  always @(posedge clock or posedge arst) begin
    if (arst) begin
      tmp107 <= 1'h0;
    end else begin
      tmp107 <= 1'h0;
    end
  end
```

closed 2022-07-03 by seldridge