llvm/circt #1523: [Calyx] Add parser for native IR representation.

cgyurgyik, 2021-08-05:
An idea discussed is allowing individuals to play with Calyx optimizations through the native compiler. 

This would require the following steps:

1. Frontend lowers to Calyx MLIR dialect.
2. Emit Calyx native IR (as seen [here](https://github.com/cucapra/calyx/blob/master/tests/correctness/if.futil)) using the `CalyxEmitter`.
3. Run passes using the native compiler and nifty driver [fud](https://capra.cs.cornell.edu/docs/calyx/fud/index.html).
4. **Parse back to Calyx MLIR dialect**.
5. Lower from Calyx to HW.

Benefits: Optimizing passes can be used without re-writing them in MLIR, a longer term goal.

rachitnigam, 2021-08-05:
This looks awesome @cgyurgyik. One other possible option is generate Calyx-MLIR format from the native Calyx compiler so that step (4) just becomes running the Calyx-MLIR parser.

cgyurgyik, 2021-09-10:
Closed. This was completed by @rachitnigam [here](https://github.com/cucapra/calyx/blob/master/src/backend/circt.rs).

closed 2021-09-10 by cgyurgyik