llvm/circt #1541: [FIRRTL] name conflict resolution mismatch

drom, 2021-08-10:
The following FIRRTL program

```python
circuit top_mod:
  module top_mod:
    input a_b : UInt<1>
    input a : { b: UInt<2> }
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input       a_b,
  input [1:0] a_b_0);

endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input        a_b,
  input  [1:0] a__b
);
endmodule
```

seldridge, 2021-08-11:
This is a fun one... For more info, Scala FIRRTL is generating new names that won't collide with anything in the module namespace. This allows the second input to "squat" on the name `a_b`.

Note: this lowering does not obey "prefix uniqueness" that the FIRRTL specification defines in section 11.2. E.g., the following circuit does not result in `a___b` (which would be a prefix unique name):

```scala
circuit top_mod:
  module top_mod:
    input a: { b: UInt<1> }
    input a_b: UInt<1>
    input a__c: UInt<1>
```

It instead porduces `a__b`:

```verilog
module top_mod(
  input   a__b,
  input   a_b,
  input   a__c
);
endmodule
```

seldridge, 2023-07-30:
The FIRRTL spec originally did not specify this. However, the spec now specifies that the MFC-style lowering is the correct one.

closed 2023-07-30 by seldridge