llvm/circt #1565: [FIRRTL] InferResets: Windows build broken

teqdruid, 2021-08-13:
https://github.com/llvm/circt/runs/3321770693?check_suite_focus=true

```
D:\a\circt\circt\lib\Dialect\FIRRTL\Transforms\InferWidths.cpp(70,21): warning C4003: not enough arguments for function-like macro invocation 'EXPR_NAMES' [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Transforms\obj.CIRCTFIRRTLTransforms.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Transforms\InferResets.cpp(1312,29): error C7555: use of designated initializers requires at least '/std:c++latest' [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Transforms\obj.CIRCTFIRRTLTransforms.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Transforms\InferWidths.cpp(1467): warning C4715: '<lambda_545279087e2b508596e8419ea1fb2b9a>::operator()': not all control paths return a value [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Transforms\obj.CIRCTFIRRTLTransforms.vcxproj]
```

teqdruid, 2021-08-13:
@fabianschuiki Looks like https://github.com/llvm/circt/commit/c3573eed43da83c1bffd2f3c8d10507ba0322e4a broke it.

seldridge, 2021-08-13:
I think just removing the designated initializers (e53f330) will fix it. Feel free to re-open if it's not (I'll know in about 6 minutes anyway).

I'm curious as to why that wasn't generating an error/warning on the clang compile for a C++20 feature. 🤔 I was used to the compiles complaining about C++17 features when we were on C++14.

closed 2021-08-13 by seldridge

fabianschuiki, 2021-08-13:
Thanks for the awesome quick fix @seldridge. That totally slipped through the cracks. Weird that it didn't get flagged... Might be a GNU-specific feature.

commit e53f330c (2021-08-13): [FIRRTL] Remove designated inits in InferResets

Remove use of designated initializers, a C++20 feature.

Fixes #1565.