llvm/circt #1568: [FIRRTL/ExportVerilog] Instance input expressions don't get emitted inline in practice

lattner, 2021-08-13:
ExportVerilog will happily emit input operands to instances inline, but we don't actually achieve this for MFC because of how FIRRTL instances work.  Consider something like:

```
circuit top_mod :
 module mod_0 :
   input input: UInt<8>

 module top_mod :
   input value: UInt<7>
   inst U1 of mod_0
   U1.input <= add(value, UInt<7>(1))
```

We currently print this to:

```
module top_mod(
  input [6:0] value);

  wire [7:0] _T;	

  mod_0 U1 (
    .input_0 (_T)
  );
  assign _T = {1'h0, value} + 8'h1;
endmodule
```

But should be able to get:

```
module top_mod(
  input [6:0] value);

  mod_0 U1 (
    .input_0 ({1'h0, value} + 8'h1)
  );
endmodule
```

The problem here is a phase ordering / cleanup issue: the input to export verilog has the add below the instance, so a wire is emitted to handle this:

```
 hw.module @top_mod(%value: i7) {
    %c1_i8 = hw.constant 1 : i8
    %false = hw.constant false
    hw.instance "U1" @mod_0(%1) : (i8) -> ()
    %0 = comb.concat %false, %value : (i1, i7) -> i8
    %1 = comb.add %0, %c1_i8 : i8
    hw.output
  }
```

One simple solution to this is to just move all instances to the bottom of a module in PrettifyVerilog, does anyone object to that?

darthscsi, 2021-08-16:
That should be fine.  Hoisting localparams above stuff and always before the instances but below other stuff at the same time would be nice.  Something like:
```
local params
var decls (first raw, then inline assignment?)
assigns
always
instances
```

lattner, 2021-08-18:
implemented in 66791e08.  We'll take the other suggestions one at a time.

closed 2021-08-18 by lattner