llvm/circt #1572: [Comb] Fold concat-with-constant into the arms of a mux that has constant operands

lattner, 2021-08-14:
This IR:

```
 hw.module @SevenSegmentDecoder(%in: i4) -> (%out: i7) {
    %c1_i4 = hw.constant 1 : i4
    %c6_i6 = hw.constant 6 : i6
    %c-1_i6 = hw.constant -1 : i6
    %false = hw.constant false
    %0 = comb.icmp eq %in, %c1_i4 : i4
    %1 = comb.mux %0, %c6_i6, %c-1_i6 : i6
    %2 = comb.concat %false, %1 : (i1, i6) -> i7
     hw.output %2 : i7
}
```

Can be simplified by folding the "concat of constant + mux" into the two arms of the mux.  This should be profitable if the mux has at least one constant operand.

fyquah, 2021-08-21:
If i understand correctly, the suggested transformation is `cat(c1, mux(cond, c2, y))`, where c1 and c2 are constants, into `mux(cond, cat(c1, c2), cat(c1, y))`. Is this definitely profitable? Won't this result in more bits of multiplexer than the original IR?

lattner, 2021-08-21:
Yes, that's what I'm suggesting, but I think we should limit it to the case when all of c1/c2/y are constant.  The end result is a mux between two constants which is more canonical than "concat{c1,mux(c2,c3)}"