llvm/circt #1573: [FIRRTL] InferResets Bug

seldridge, 2021-08-14:
The `InferResets` pass is not inferring a reset involving a bundle in the following circuit.  I'm not sure what's going on.

```scala
circuit Foo:
  module Bar:
    input a : {b : {reset : Reset}}

  module Foo :
    input reset : UInt<1>

    inst bar of Bar

    bar.a.b.reset <= asAsyncReset(UInt<1>("h01"))
```

Running this with `firtool Foo.fir --debug` produces:

```
===----- Tracing uninferred resets -----===

Visiting instance bar
Visiting driver 'a.b.reset' = 'bar.a.b.reset' (!firrtl.reset = !firrtl.reset)
Visiting driver '' = 'reset' (!firrtl.reset = !firrtl.reset)
Visiting driver '' = '' (!firrtl.reset = !firrtl.asyncreset)

===----- Infer reset types -----===

Inferring reset network with 2 nodes
- Found 0 async, 0 sync, 0 invalid drives
Reduced.fir:8:5: error: reset network never driven with concrete type
    inst bar of Bar
    ^
```

Interestingly, this works fine if the type is: `input a: {reset: Reset}`.

youngar, 2021-08-14:
attempt at a fix: https://github.com/llvm/circt/pull/1576

fabianschuiki, 2021-08-16:
That fix looks great, thanks @youngar :100:!

closed 2021-08-16 by youngar