llvm/circt #1575: [Calyx] Add memory primitive

mikeurbach, 2021-08-14:
The Calyx standard library includes memories: https://capra.cs.cornell.edu/docs/calyx/libraries/core.html#memories.

Would it make sense to add this alongside the register primitive in CIRCT? https://github.com/llvm/circt/blob/main/include/circt/Dialect/Calyx/CalyxPrimitives.td

I've been playing around with some conversions that would make use of the memory abstraction if it existed.

cgyurgyik, 2021-08-14:
Each primitive in the Calyx primitives library entails the following: 
1. An interface to define ports, attributes, and (optional) parameters.
2. A component which implements the given interface in SV.

Let's look at an example, `std_add` which is an adder.

First, we have the [interface](https://github.com/cucapra/calyx/blob/40ec4e8d5acfad98d7b79ad1e280fa47ce591401/primitives/core.futil#L14) defined in `core.futil`:

```
extern "core.sv" {
  primitive std_add<"share"=1>[WIDTH](left: WIDTH, right: WIDTH) -> (out: WIDTH);
}
```

And then we have the [definition](https://github.com/cucapra/calyx/blob/40ec4e8d5acfad98d7b79ad1e280fa47ce591401/primitives/core.sv#L102-L110) in `core.sv`:

```
module std_add #(
    parameter WIDTH = 32
) (
   input wire               logic [WIDTH-1:0] left,
   input wire               logic [WIDTH-1:0] right,
   output logic [WIDTH-1:0] out
);
  assign out = left + right;
endmodule
```

In a Calyx program, we'd instantiate an instance of this in the following manner:
```
component main () -> () {
  cells {
    my_adder = std_add(/*width=*/32);
  }
  wires {
    // Now, we can assign values to my_adder.left, my_adder.right
  }
  ...
}
```

A question that remains unanswered is how to cleanly translate this to the Calyx dialect. A reasonable approach IMO is to define each of these as an operation, and then provide their SV implementations somewhere for lowering to HW.

mikeurbach, 2021-08-14:
> A reasonable approach IMO is to define each of these as an operation

This is what you did for RegisterOp right? I think that makes sense.

> then provide their SV implementations somewhere for lowering to HW

As in, a System Verilog file that the lowering to HW can reference through external modules? That's what some of the ESI [stuff](https://github.com/llvm/circt/blob/main/include/circt/Dialect/ESI/ESIPrimitives.sv) has done, and it could make sense here. Another approach might be to define RewritePatterns to convert them to HW/Comb/Seq/SV dialects. That would be a bit more effort than just writing out the System Verilog, but those patterns could be applied during lowering to HW, and they could also potentially be useful to other passes. IIRC ESI would have taken that approach, but we lacked the necessary Seq/SV constructs at the time.

cgyurgyik, 2021-08-15:
Looking to provide a better Cell interface in #1577 (WIP), that also unifies primitives and instances of components.

mikeurbach, 2021-08-16:
I think that makes a lot of sense. I prototyped a MemoryOp primitive and it would work very nicely on top of the CalyxPrimitive you're adding in the above PR.

mikeurbach, 2021-08-25:
Closed by #1632

closed 2021-08-25 by mikeurbach