llvm/circt #1578: [FIRRTL] InferReset failing to update extmodules

youngar, 2021-08-15:
InferTesets should be updating the reset type on external modules.

`firtool ./infer.fir`
```
circuit bar:
  extmodule X:
    input reset : Reset

  module bar:
    inst x of X
    x.reset <= UInt<1>("h1")
 ```

```
/work/aryoung/infer.fir:10:5: error: 'firrtl.instance' op result type for "reset" must be '!firrtl.reset', but got '!firrtl.uint<1>'
    inst x of X
    ^
/work/aryoung/infer.fir:10:5: note: see current operation: %0 = "firrtl.instance"() {annotations = [], lowerToBind = false, moduleName = @X, name = "x", portAnnotations = [[]]} : () -> !firrtl.uint<1>
/work/aryoung/infer.fir:6:13: note: original module declared here
  extmodule X:
```

fabianschuiki, 2021-08-16:
I was wondering about this. Intuitively since extmodules are external and we cannot influence their interface, reset inference should not touch these. But then again, since `asyncreset` and `uint<1>` resolve to the same type in Verilog, this should probably just be updated as expected.

closed 2021-08-16 by fabianschuiki

seldridge, 2021-08-16:
I don't know why this is the case, but SFC will do it... E.g., SFC converts:

```scala
circuit Foo:
  extmodule Bar:
    input a: Reset
    input b: Reset
  module Foo:
    input a: UInt<1>
    input b: AsyncReset

    inst bar of Bar
    bar.a <= a
    bar.b <= b
```

Into:

```scala
circuit Foo :
  extmodule Bar :
    input a : UInt<1>
    input b : AsyncReset
    defname = Bar

  module Foo :
    input a : UInt<1>
    input b : AsyncReset

    inst bar of Bar
    bar.a <= a
    bar.b <= b
```

Thanks for the fix!

fabianschuiki, 2021-08-16:
I guess it makes sense in order to get the types to line up. But arguably having a `ResetType` go into an extmodule -- whose interface is supposed to be fixed -- does sound like a design smell / linting opportunity 🤔

commit 0afd84d2 (2021-08-16): [InferResets] Update extmodule types

Also update the types of extmodule ports that were inferred during reset
inference. In general we do not want to update the type of extmodules,
since these are provided from the outside and the interface is thus
fixed. However in the case of reset inference where the types all map to
the same Verilog, this is okay to do.

This currently uses the types inferred for `InstanceOp`s to determine
what types the corresponding extmodule should have. This works in the
cases we are interested in, but can break in some corner cases where a
reset network is only conne [... cut]