llvm/circt #1579: [FIRRTL] Infer reset having trouble with `invalidvalue`

youngar, 2021-08-15:
InferResets is treating invalid values of `reset` type as regular values which need to be inferred in a reset network.  This gets very nasty when InferResets runs after CSE and all `%invalid_reset = firrtl.invalidvalue : !firrtl.reset` values get combined in to a single value.

`./bin/firtool --print-ir-before-all --mlir-disable-threading  /work/aryoung/infer.fir`

```
circuit foo:
  module foo:
    input reset : UInt<1>
    output r0 : Reset
    output r1 : Reset

    r0 is invalid
    r1 is invalid

    node _t = asAsyncReset(reset)

    r0 <= reset
    r1 <= _t
```

```mlir
// -----// IR Dump Before CSE //----- //
firrtl.module @foo(in %reset: !firrtl.uint<1>, out %r0: !firrtl.reset, out %r1: !firrtl.reset) {
  %invalid_reset = firrtl.invalidvalue : !firrtl.reset
  firrtl.connect %r0, %invalid_reset : !firrtl.reset, !firrtl.reset
  %invalid_reset_0 = firrtl.invalidvalue : !firrtl.reset
  firrtl.connect %r1, %invalid_reset_0 : !firrtl.reset, !firrtl.reset
  %0 = firrtl.asAsyncReset %reset : (!firrtl.uint<1>) -> !firrtl.asyncreset
  %_t = firrtl.node %0  : !firrtl.asyncreset
  firrtl.connect %r0, %reset : !firrtl.reset, !firrtl.uint<1>
  firrtl.connect %r1, %_t : !firrtl.reset, !firrtl.asyncreset
}
```

```mlir
// -----// IR Dump Before InferResets //----- //
firrtl.circuit "foo"   {
  firrtl.module @foo(in %reset: !firrtl.uint<1>, out %r0: !firrtl.reset, out %r1: !firrtl.reset) {
    %invalid_reset = firrtl.invalidvalue : !firrtl.reset
    firrtl.connect %r0, %invalid_reset : !firrtl.reset, !firrtl.reset
    firrtl.connect %r1, %invalid_reset : !firrtl.reset, !firrtl.reset
    %0 = firrtl.asAsyncReset %reset : (!firrtl.uint<1>) -> !firrtl.asyncreset
    %_t = firrtl.node %0  : !firrtl.asyncreset
    firrtl.connect %r0, %reset : !firrtl.reset, !firrtl.uint<1>
    firrtl.connect %r1, %_t : !firrtl.reset, !firrtl.asyncreset
  }
}
```
This is a hacked up error message different from the default one:
```
/work/aryoung/infer.fir:3:10: error: reset network simultaneously connected to async and sync resets
  module foo:
         ^
/work/aryoung/infer.fir:3:10: note: did you intend for the reset to be async?
/work/aryoung/infer.fir:8:8: note: offending
dest is sync
src is sync
 drive here:
    r0 is invalid
       ^
/work/aryoung/infer.fir:9:8: note: offending
dest is sync
src is sync
 drive here:
    r1 is invalid
       ^
/work/aryoung/infer.fir:13:8: note: offending
dest is sync
src is sync
 drive here:
    r0 <= reset
       ^
/work/aryoung/infer.fir:14:8: note: offending
dest is sync
src is async
 drive here:
    r1 <= _t
```

Ramlakshmi3733, 2021-08-15:
I am puzzled by the output of the Scala Compiler as well.  It seems to ignore the asAsyncReset statement of the example:
module foo(
  input   reset,
  output  r0,
  output  r1
);
  assign r0 = reset;
  assign r1 = reset;
endmodule

fabianschuiki, 2021-08-16:
This is an excellent catch! Indeed the `invalidvalue` op here causes the networks to join along the common op, which is incorrect. I pushed a fix in ac94c63 and added a test for this. The breaking example you have posted is now working.

fabianschuiki, 2021-08-16:
Marking this as fixed in ac94c63.

closed 2021-08-16 by fabianschuiki