llvm/circt #1580: [FIRRTL][ExportVerilog] Integration test `EmitVerilog/exampleCodeGen.fir` broken

teqdruid, 2021-08-15:
Appears to have been broken by one of the commits on Friday.

https://github.com/llvm/circt/runs/3328956237?check_suite_focus=true

```
FAIL: CIRCT :: EmitVerilog/exampleCodeGen.fir (27 of 27)
******************** TEST 'CIRCT :: EmitVerilog/exampleCodeGen.fir' FAILED ********************
Script:
--
: 'RUN: at line 2';   /__w/circt/circt/build/bin/firtool -verilog -lower-to-hw /__w/circt/circt/integration_test/EmitVerilog/exampleCodeGen.fir > /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/exampleCodeGen.fir.tmp.sv
: 'RUN: at line 3';   /__w/circt/circt/build/bin/circt-rtl-sim.py /__w/circt/circt/build/tools/circt/integration_test/EmitVerilog/Output/exampleCodeGen.fir.tmp.sv 2>&1 | grep PASS
--
Exit Code: 1
```

Bug could be in `ExportVerilog` or FIRRTL somewhere.

teqdruid, 2021-08-15:
@lattner : `git bisect` shows that df37fa11c776879080b681b38653617c9f91e9c2 (`[Comb] Fold a large series of of mux ops into array_create/get (#1559)`) is the first bad commit.

lattner, 2021-08-15:
Thanks for noticing this, I'll take a look!

teqdruid, 2021-08-15:
I get email notifications, which makes it easy to notice ;)

lattner, 2021-08-16:
Do you have any idea what is [going on here](https://github.com/llvm/circt/actions/runs/1135378285)?  All the failures look like "something was canceled"?

teqdruid, 2021-08-16:
When one job fails, the rest get cancelled. One of them got to the tests
and failed the tests. Unfortunately, the output (which I posted above)
isn't too helpful.

lattner, 2021-08-16:
I changed the test last night to produce more useful output.  Do you have a link to where the more recent failure prints out its output?  I'm having trouble understanding the github interface

teqdruid, 2021-08-16:
[InferResets] Update extmodule types · ***@***.*** (github.com)
<https://github.com/llvm/circt/runs/3341979376?check_suite_focus=true>

Just click on the red "integration tests" badge below the file list. (This
is a shortcut to the failing workflow.)

lattner, 2021-08-16:
@fabianschuiki can you take a look at this, it looks like your patch broke an already broken builder

fabianschuiki, 2021-08-16:
Huh interesting. I'll take a look. Having a `ResetType` in the ports of an extmodule should be the only case affected by the patch, and that use case needs some further clarification as to whether it actully makes sense. Wonder what triggered this here.

lattner, 2021-08-16:
It looks like I got the endianness wrong on array_create.

teqdruid, 2021-08-16:
That's bitten me a few times before.

lattner, 2021-08-16:
Should be fixed in c6fd4f1b thank you for pinging me on this!

lattner, 2021-08-16:
btw, the generated verilog on that test case is way way better.

closed 2021-08-16 by lattner