llvm/circt #1583: [FIRRTL] ExpandWhens generates unnecessary NotPrimOp

seldridge, 2021-08-16:
Chisel's `when` and `switch` APIs can result in large, nested mux trees. While the canonicalization done in https://github.com/llvm/circt/commit/ee457605169c16b1ce1400201e66d06d723f282b is amazing, this isn't catching situations involving `switch` or `when` because of how `ExpandWhens` is lowering `when`/`else` structures to include a not checking that the outer condition isn't true. 

Consider something like:

```scala
circuit Foo:
  module Foo:
    input sel: UInt<2>
    input a_0: UInt<1>
    input a_1: UInt<1>
    input a_2: UInt<1>
    input a_3: UInt<1>
    output b: UInt<1>

    b <= a_0
    when eq(sel, UInt<2>(0)) :
      b <= a_1
    else when eq(sel, UInt<2>(1)) :
      b <= a_2
    else when eq(sel, UInt<2>(2)) :
      b <= a_3
```

`ExpandWhens` is lowering this to:

```mlir
firrtl.module @Foo(in %sel: !firrtl.uint<2>, in %a_0: !firrtl.uint<1>, in %a_1: !firrtl.uint<1>, in %a_2: !firrtl.uint<1>, in %a_3: !firrtl.uint<1>, out %b: !firrtl.uint<1>) {
  %c0_ui2 = firrtl.constant 0 : !firrtl.uint<2>
  %0 = firrtl.eq %sel, %c0_ui2 : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint<1>
  %c1_ui2 = firrtl.constant 1 : !firrtl.uint<2>
  %c2_ui2 = firrtl.constant 2 : !firrtl.uint<2>
  %1 = firrtl.not %0 : (!firrtl.uint<1>) -> !firrtl.uint<1>
  %2 = firrtl.eq %sel, %c1_ui2 : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint<1>
  %3 = firrtl.and %1, %2 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  %4 = firrtl.not %2 : (!firrtl.uint<1>) -> !firrtl.uint<1>
  %5 = firrtl.and %1, %4 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  %6 = firrtl.eq %sel, %c2_ui2 : (!firrtl.uint<2>, !firrtl.uint<2>) -> !firrtl.uint<1>
  %7 = firrtl.and %5, %6 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  %8 = firrtl.mux(%3, %a_2, %a_3) : (!firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  %9 = firrtl.mux(%0, %a_1, %8) : (!firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
  firrtl.connect %b, %9 : !firrtl.uint<1>, !firrtl.uint<1>
}
```

Notably, each of the mux conditions (after the outermost) include a check that the condition is true and the previous condition is not true.

Comparing this with what the Scala FIRRTL Compiler does here to produce, after it's `ExpandWhens` pass:

```scala
circuit Foo :
  module Foo :
    input sel : UInt<2>
    input a_0 : UInt<1>
    input a_1 : UInt<1>
    input a_2 : UInt<1>
    input a_3 : UInt<1>
    output b : UInt<1>

    node _GEN_0 = mux(eq(sel, UInt<2>("h2")), a_3, a_0)
    node _GEN_1 = mux(eq(sel, UInt<2>("h1")), a_2, _GEN_0)
    node _GEN_2 = mux(eq(sel, UInt<2>("h0")), a_1, _GEN_1)
    b <= _GEN_2
```

The end result of this is that these extra `and(cond, not(old_cond))` produce code which ee457605169c16b1ce1400201e66d06d723f282b isn't sophisticated enough to canonicalize away. E.g., using CIRCT, you will get some obnoxious mux trees in the output that look like:

```verilog
module Foo(
  input  [1:0] sel,
  input        a_0, a_1, a_2, a_3,
  output       b);

  assign b = |sel ? (|sel & sel == 2'h1 ? a_2 : a_3) : a_1;	// Tmp.fir:2:10, :11:{5,10}, :13:{15,23}, :16:9
endmodule
```

I'd expect to get the nice array structure.

For a larger example, the following should compile to the mux tree in #675:

```scala
circuit SevenSegmentDecoderWhen :
  module SevenSegmentDecoderWhen :
    input in : UInt<4>
    output out : UInt<7>

    node _T = eq(in, UInt<1>("h0"))
    when _T :
      out <= UInt<6>("h3f")
    else :
      node _T_1 = eq(in, UInt<1>("h1"))
      when _T_1 :
        out <= UInt<3>("h6")
      else :
        node _T_2 = eq(in, UInt<2>("h2"))
        when _T_2 :
          out <= UInt<7>("h5b")
        else :
          node _T_3 = eq(in, UInt<2>("h3"))
          when _T_3 :
            out <= UInt<7>("h4f")
          else :
            node _T_4 = eq(in, UInt<3>("h4"))
            when _T_4 :
              out <= UInt<7>("h66")
            else :
              node _T_5 = eq(in, UInt<3>("h5"))
              when _T_5 :
                out <= UInt<7>("h6d")
              else :
                node _T_6 = eq(in, UInt<3>("h6"))
                when _T_6 :
                  out <= UInt<7>("h7d")
                else :
                  node _T_7 = eq(in, UInt<3>("h7"))
                  when _T_7 :
                    out <= UInt<3>("h7")
                  else :
                    node _T_8 = eq(in, UInt<4>("h8"))
                    when _T_8 :
                      out <= UInt<7>("h7f")
                    else :
                      node _T_9 = eq(in, UInt<4>("h9"))
                      when _T_9 :
                        out <= UInt<7>("h6f")
                      else :
                        node _T_10 = eq(in, UInt<4>("ha"))
                        when _T_10 :
                          out <= UInt<7>("h77")
                        else :
                          node _T_11 = eq(in, UInt<4>("hb"))
                          when _T_11 :
                            out <= UInt<7>("h7c")
                          else :
                            node _T_12 = eq(in, UInt<4>("hc"))
                            when _T_12 :
                              out <= UInt<6>("h39")
                            else :
                              node _T_13 = eq(in, UInt<4>("hd"))
                              when _T_13 :
                                out <= UInt<7>("h5e")
                              else :
                                node _T_14 = eq(in, UInt<4>("he"))
                                when _T_14 :
                                  out <= UInt<7>("h79")
                                else :
                                  node _T_15 = eq(in, UInt<4>("hf"))
                                  when _T_15 :
                                    out <= UInt<7>("h71")
                                  else :
                                    out <= UInt<1>("h0")
```

youngar, 2021-08-16:
With: https://github.com/llvm/circt/pull/1585 the two examples are now:
```verilog
module Foo(
  input  [1:0] sel,
  input        a_0, a_1, a_2, a_3,
  output       b);

  assign b = sel == 2'h0 ? a_1 : sel == 2'h1 ? a_2 : a_3;	// ./ew.fir:2:10, :11:10, :13:{15,23}, :16:9
endmodule
```

```verilog
module SevenSegmentDecoderSwitch(
  input  [3:0] in,
  output [6:0] out);

  wire [15:0][6:0] _tmp = {{7'h3F}, {7'h6}, {7'h5B}, {7'h4F}, {7'h66}, {7'h6D}, {7'h7D}, {7'h7}, {7'h7F}, {7'h6F}, {7'h77}, {7'h7C}, {7'h39}, {7'h5E}, {7'h79}, {7'h71}};	// ./ssd.fir:11:19, :15:21, :17:18, :19:23, :21:20, :23:25, :25:22, :27:27, :29:24, :31:29, :33:26, :35:31, :39:33, :41:30, :43:35, :45:32, :47:38, :49:34, :51:40, :53:36, :55:42, :59:44, :61:40, :63:46, :65:42, :69:{41,44}
  assign out = _tmp[in];	// ./ssd.fir:2:10, :17:18, :21:20, :25:22, :29:24, :33:26, :41:30, :45:32, :49:34, :53:36, :61:40, :65:42, :69:{41,44}
endmodule
```

closed 2021-08-16 by youngar

lattner, 2021-08-17:
nice!