llvm/circt #1586: [Python Bindings] Windows cmake: `multiple rules generate lib/_mlir.lib`

teqdruid, 2021-08-16:
When trying to run a unified build on Windows which includes the CIRCT Python bindings, I get:

```
> ninja.exe check-pycde
ninja: error: build.ninja:99751: multiple rules generate lib/_mlir.lib [-w dupbuild=err]
```

Here is my configure command (run through VSCode, which is why it looks so explicit):
```
"C:\Program Files\CMake\bin\cmake.EXE" --no-warn-unused-cli -DLLVM_ENABLE_PROJECTS:STRING=mlir -DCMAKE_BUILD_TYPE:STRING=Debug -DLLVM_TARGETS_TO_BUILD:STRING=X86 -DLLVM_ENABLE_ASSERTIONS:STRING=ON -DMLIR_ENABLE_BINDINGS_PYTHON:STRING=ON -DLLVM_EXTERNAL_PROJECTS:STRING=circt -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR:STRING=c:/code/circt -DCIRCT_BINDINGS_PYTHON_ENABLED:STRING=ON -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -Hc:/code/circt/llvm/llvm -Bc:/code/circt/build -G Ninja
```

I have yet to look through the cmake files myself. Just posting this here so as not to lose track of it and to see if others have seen this.

teqdruid, 2021-08-16:
@mikeurbach @stellaraccident Have you seen this before?

mikeurbach, 2021-08-16:
I haven't, and just double checked on main. I'm on Ubuntu 20.04 using CMake 3.19.7 and Ninja 1.10.0.

Just sanity check, is this on a clean build?

teqdruid, 2021-08-16:
I re-configured in a fresh directory, so yes. The build doesn't even run as this is an error (albeit ignorable with a flag). Ninja does this for every target, presumably as it does this check before it even gets the list of targets:

```
> ninja
ninja: error: build.ninja:99751: multiple rules generate lib/_mlir.lib [-w dupbuild=err]
```

I'm using Windows 10, cmake 3.19.3, and ninja 1.10.2.

mikeurbach, 2021-08-16:
I'm less familiar with the Windows build, and what is going on with that target. On Linux the `_mlir.<platform>.so` lib is being built in `build/tools/mlir/python_packages/mlir_core/mlir/_mlir_libs/`.

teqdruid, 2021-08-16:
> I'm less familiar with the Windows build

That makes two of us!

stellaraccident, 2021-08-16:
Sadly, i know things about windows. Wish I didn't.

Windows shared libraries output both a dll and a lib file. The paths to both must be overridden if building to a non standard location in the tree (which python extensions are). Otherwise, the lib file goes in the lib/ directory and the dll goes in the bin directory. The issue here is that two things are trying to generate a _mlir.dll so into their own build directories (which is fine), but since the lib directory is not being set, they are both trying to write that file to the shared, project wide lib directory.

stellaraccident, 2021-08-16:
The issue has to be originating round about here: https://github.com/llvm/llvm-project/blob/b2aaafb8377ac1ab081e7c0d3ba92ee5eb4de07c/mlir/cmake/modules/AddMLIRPython.cmake#L482

I need to review some cmake docs for the subtlety on the windows side.

teqdruid, 2021-08-16:
> Sadly, i know things about windows. Wish I didn't.

Sadly, I have to know things about Windows, but don't. Wish I could remain blissfully ignorant.

stellaraccident, 2021-08-16:
https://cmake.org/cmake/help/latest/manual/cmake-buildsystem.7.html#archive-output-artifacts

Ok, these are considered "archive files". Should be a one line change, in the WIN32 branch of the above, also set ARCHIVE_OUTPUT_LOCATION in addition to `RUNTIME_OUTPUT_DIRECTORY ${ARG_OUTPUT_DIRECTORY}`

teqdruid, 2021-08-16:
Excellent! I'll try that and if it works I'll submit a patch upstream. Thanks!

teqdruid, 2021-08-17:
Fixed upstream: https://github.com/llvm/llvm-project/commit/5821047aaca23184ee30fb212bfbbfa25967448b

Thanks @stellaraccident !

stellaraccident, 2021-08-17:
I'm glad my misspent youth in the bowels of the windows stack continues to help folks :)

teqdruid, 2021-08-17:
Next hurdle: just like the MLIR python tests, the CIRCT tests can't find the `_mlir.dll`. From `check-mlir`:

```
Traceback (most recent call last):
  File "C:\code\circt\llvm\mlir\test\python\dialects\shape.py", line 3, in <module>
    from mlir.ir import *
  File "c:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\ir.py", line 6, in <module>
    from ._cext_loader import _reexport_cext
  File "c:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_cext_loader.py", line 34, in <module>
    _cext = load_extension("_mlir")
  File "c:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs\__init__.py", line 18, in load_extension
    return importlib.import_module(f".{name}", __package__)
  File "c:\python39\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed while importing _mlir: The specified module could not be found.
```

Indeed, MSVC doesn't appear to be creating `_mlir.dll`. Any hints?

stellaraccident, 2021-08-17:
> Indeed, MSVC doesn't appear to be creating _mlir.dll. Any hints?

Python extensions in windows are named with the .pyd extension, not. dll. This is configured here: https://github.com/llvm/llvm-project/blob/8f750e881414ed26d10ad01f66d0d50d4807070e/mlir/cmake/modules/AddMLIRPython.cmake#L487

Assuming it exists, the next thing to verify is whether it actually can't find the module or whether it just can't load the dll (typically due to a dependency that can't be found). My bet is on the latter. We need a bit more elaborate module loader to do this right. However, if you set your PATH to include the _mlir_libs directory, it should be able to find the dependencies -- and then you'll know that is the issue.

stellaraccident, 2021-08-17:
We ultimately need some platform specific code which preloads the CAPI dll before loading a native extension. Then the subsequent load of the extension will find the dependent dll by name vs searching the path.

This is all because windows lacks an equivalent to the rpath on Linux and co.

stellaraccident, 2021-08-17:
Example from a prior try at this: https://github.com/llvm/llvm-project/blob/8f750e881414ed26d10ad01f66d0d50d4807070e/mlir/python/mlir/_dlloader.py#L58

teqdruid, 2021-08-17:
> Python extensions in windows are named with the .pyd extension, not. dll. This is configured here: https://github.com/llvm/llvm-project/blob/8f750e881414ed26d10ad01f66d0d50d4807070e/mlir/cmake/modules/AddMLIRPython.cmake#L487

Ah. This makes sense and this issue now jives with your previous Discord comment about finding it.

> Assuming it exists

```
> ls C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs

    Directory: C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d----           8/16/2021  8:15 PM                __pycache__
d----           8/16/2021  7:46 PM                include
la---           8/17/2021  4:42 PM              0 __init__.py -> C:\code\circt\llvm\mlir\python\mlir\_mlir_libs\__init__.py
-a---           8/17/2021  3:49 PM       74065112 _mlir.cp39-win_amd64.ilk
-a---           8/17/2021  3:49 PM        7589376 _mlir.cp39-win_amd64.pyd
-a---           8/16/2021  8:14 PM            713 _mlir.exp
-a---           8/16/2021  8:14 PM           1964 _mlir.lib
-a---           8/17/2021  3:49 PM       53112832 _mlir.pdb
....
```

So it exists. As for whether or not Python is finding it or can't load it:

```
> echo $env:PATH
[...];C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs;C:\code\circt\build\bin

> $env:PYTHONPATH="C:\code\circt\build\tools\mlir\python_packages\mlir_core"

> python C:\code\circt\llvm\mlir\test\python\ir\affine_expr.py
Traceback (most recent call last):
  File "C:\code\circt\llvm\mlir\test\python\ir\affine_expr.py", line 4, in <module>
    from mlir.ir import *
  File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\ir.py", line 6, in <module>
    from ._cext_loader import _reexport_cext
  File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_cext_loader.py", line 34, in <module>
    _cext = load_extension("_mlir")
  File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs\__init__.py", line 18, in load_extension
    return importlib.import_module(f".{name}", __package__)
  File "C:\Python39\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed while importing _mlir: The specified module could not be found.
```

Though I suspect that's what you expected. I'll step through `_dllloader.py` & experiment later.

teqdruid, 2021-08-17:
> > Python extensions in windows are named with the .pyd extension, not. dll. This is configured here: https://github.com/llvm/llvm-project/blob/8f750e881414ed26d10ad01f66d0d50d4807070e/mlir/cmake/modules/AddMLIRPython.cmake#L487
> 
> Ah. This makes sense and this issue now jives with your previous Discord comment about finding it.
> 
> > Assuming it exists
> 
> ```
> > ls C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs
> 
>     Directory: C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs
> 
> Mode                 LastWriteTime         Length Name
> ----                 -------------         ------ ----
> d----           8/16/2021  8:15 PM                __pycache__
> d----           8/16/2021  7:46 PM                include
> la---           8/17/2021  4:42 PM              0 __init__.py -> C:\code\circt\llvm\mlir\python\mlir\_mlir_libs\__init__.py
> -a---           8/17/2021  3:49 PM       74065112 _mlir.cp39-win_amd64.ilk
> -a---           8/17/2021  3:49 PM        7589376 _mlir.cp39-win_amd64.pyd
> -a---           8/16/2021  8:14 PM            713 _mlir.exp
> -a---           8/16/2021  8:14 PM           1964 _mlir.lib
> -a---           8/17/2021  3:49 PM       53112832 _mlir.pdb
> ....
> ```
> 
> So it exists. As for whether or not Python is finding it or can't load it:
> 
> ```
> > echo $env:PATH
> [...];C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs;C:\code\circt\build\bin
> 
> > $env:PYTHONPATH="C:\code\circt\build\tools\mlir\python_packages\mlir_core"
> 
> > python C:\code\circt\llvm\mlir\test\python\ir\affine_expr.py
> Traceback (most recent call last):
>   File "C:\code\circt\llvm\mlir\test\python\ir\affine_expr.py", line 4, in <module>
>     from mlir.ir import *
>   File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\ir.py", line 6, in <module>
>     from ._cext_loader import _reexport_cext
>   File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_cext_loader.py", line 34, in <module>
>     _cext = load_extension("_mlir")
>   File "C:\code\circt\build\tools\mlir\python_packages\mlir_core\mlir\_mlir_libs\__init__.py", line 18, in load_extension
>     return importlib.import_module(f".{name}", __package__)
>   File "C:\Python39\lib\importlib\__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ImportError: DLL load failed while importing _mlir: The specified module could not be found.
> ```
> 
> Though I suspect that's what you expected. I'll step through `_cext_loader.py` and experiment with `_dllloader.py` later.

teqdruid, 2021-08-18:
When I copy `MLIRPythonCAPI.dll` to `_mlir_libs` (alongside `_mlir.cp39-win_amd64.pyd`), I get this:

[image]

So it appears that Python wasn't finding `MLIRPythonCAPI.dll`. Now the only question in my mind is whether or not that assertion failure is due to a code bug or is still linking related.

stellaraccident, 2021-08-18:
It is usually best to try a simple standalone thing when debugging

```
import mlir._mlir_libs._mlir
```

And as some other prints like `sys.path` and `on.environ["PATH"]` (and also print the python version and make sure it matches the extension version). You're solidly in debugging territory now. I suspect the problem is that it can't find the dependent CAPI dll (verify with depends.exe).

stellaraccident, 2021-08-18:
That's new to me. If I was throwing darts, I'd suspect one of those words runtime library mismatches that happen on windows (debug vs release vs single threaded vs multi threaded vs whatever).

teqdruid, 2021-08-18:
> That's new to me. If I was throwing darts, I'd suspect one of those weird runtime library mismatches that happen on windows (debug vs release vs single threaded vs multi threaded vs whatever).

<sigh>

Yeah, I'll look at that.

teqdruid, 2021-08-18:
Thanks much for the quick help!

stellaraccident, 2021-08-18:
Np. I strongly suspect that these are mostly plain dll linking/loading issues. Treating the .pyd and your dll files just as regular things and using the MSFT tools to do "dll things" with them usually yields clues.

teqdruid, 2021-09-03:
@stellaraccident I just tried LLVM main (5f6f33da9ee6cbaef5f10b4a7be34a91d5185b2b) which includes your two latest patches and I still get the python lib assertion above. I'm not compiling CIRCT in the build -- just the LLVM repo. I used https://github.com/lucasg/Dependencies to check for missing deps. I saw a few in Windows DLLs, but nothing stood out. @sstamenova volunteered to help try to repro my issue if I don't have success, but before that I just want to make sure I'm not doing something really stupid.

I'd like to repro your environment. What are you using?

cmake version 3.19.3
Visual Studio 2019 Version 16.11.1
ninja 1.10.2
cmake configure:
```
    "LLVM_ENABLE_PROJECTS": "mlir",
    "CMAKE_BUILD_TYPE": "Release",
    "LLVM_TARGETS_TO_BUILD": "X86",
    "BUILD_SHARED_LIBS": "OFF",
    "LLVM_ENABLE_ASSERTIONS": "ON",
    "MLIR_ENABLE_BINDINGS_PYTHON": "ON",
    "CMAKE_INSTALL_PREFIX": "${workspaceFolder}/build/install",
```
With the "Visual Studio Enterprise 2019 Release - amd64" toolchain (VSCode).
Windows 10 Enterprise 21H1
Python 3.9.6 (tags/v3.9.6:db3ff76, Jun 28 2021, 15:26:21) [MSC v.1929 64 bit (AMD64)] on win32

stellaraccident, 2021-09-04:
Unfortunately, my windows workstation is at home and I'm not there. I can
check later this weekend probably. Looks similar to my setup but I probably
haven't updated python in a long while (3.8 I think), and have no idea
where it came from originally. That computer is for games and occasional
LLVM sanity checking :)

teqdruid, 2021-09-04:
No rush. Getting CIRCT working on windows has been and continues to be a slow burn.

teqdruid, 2021-09-04:
Aha! It works with Python 3.8 so long as it is the only version of Python installed! Three of the tests fail, but it doesn't assert.

The failing tests:
```
Failed Tests (3):
  MLIR :: python/dialects/shape.py
  MLIR :: python/dialects/sparse_tensor/test_SpMM.py
  MLIR :: python/execution_engine.py
```

I'll try Python 3.9.7 as the only version installed next.

teqdruid, 2021-09-04:
3.9.7 worked on its own. Then it worked with 3.8.10 installed. (Both times with the above errors.) The difference between now and what I was doing previously seems to be cmake configure running from VSCode vs. running from a terminal (both compiling on the terminal). When I tried compiling with just 3.9.7 installed and configuring from VSCode, the build had a linking error since it couldn't find `python39_d.dll`! I'm thinking that when running configure from VSCode it got confused and was linking against the wrong dll. Or something like that.

Either way, this is working now (modulo the three MLIR tests). Thanks for the help @stellaraccident !

closed 2021-09-04 by teqdruid

stellaraccident, 2021-09-04:
Weird but also sounds reminiscent of issues that pop up on macos sometimes: cmake autoconfigure of python include dir, flags, and library is... Baroque. And easily gets the wrong thing sometimes based on environment. This is why in setup.py, I query python myself for what it knows and pass python configure flags manually (since that happens "in the field" and needs to be robust).

I've gone to file cmake bugs on this a couple of times but it is hard to nail down enough even to write a good bug report.

If you squint at all of the cmake output or look through the cache, you can usually spot when it has mixed and matched settings from two installed pythons, which is what this sounds like.

Also, I never invoke cmake from an ide. Call me old fashioned :)