llvm/circt #1587: [FIRRTL][LowerToHW] Mux Tree Lowering Producing Busted Verilog

seldridge, 2021-08-16:
The mux tree lowering that was added in ee457605169c16b1ce1400201e66d06d723f282b is producing some invalid Verilog in certain situations.

Consider the following FIRRTL IR:

```scala
circuit Foo:
  module Foo:
    input idx: UInt<2>
    input a: UInt<4>[4]
    output b: UInt<4>

    b <= a[idx]
```

This is currently being lowered to Verilog as:

```verilog
module Foo(
  input  [1:0] idx,
  input  [3:0] a_0, a_1, a_2, a_3,
  output [3:0] b);

  assign b = ({{a_0}, {a_1}, {a_2}, {a_3}})[idx];	// Tmp.fir:2:10, :7:12
endmodule
```

The problem is that the wire associated with `{{a_0}, {a_1}, ...}` cannot be inlined as this becomes illegal Verilog.  This can be hit for situations where the mux-encoded wire is small enough to be inlined.  The problem doesn't show up for situations where the mux-encoded wire is getting spilled.

Concretely, the above has to emit as:

```verilog
wire bar [3:0][3:0] = {{a_0}, {a_1}, ...};
assign b = bar[idx];
```

The other two inlined approaches both are wrong. 

The following is trying to bit extract out of a concatenation:

```verilog
assign b = {{a_0}, {a_1}, ...}[idx];
```

The following is just illegal:

```verilog
assign b = ({{a_0}, {a_1}, ...})[idx];
```

closed 2021-08-16 by seldridge

lattner, 2021-08-17:
Thanks for fixing this @seldridge.  Should we emit all array_create's out of line?

seldridge, 2021-08-17:
It's probably not necessary.  (And I'm not totally sure...) What I'm planning to do is to turn on Verilator linting of the `hw-dialect.mlir` and `sv-dialect.mlir` tests so that we will know immediately if we start doing something weird and need to block inlining in more situations.

It doesn't look like we currently have it, but I don't see a problem with doing `hw.array_create` and then having a `hw.cast` which does inline.  E.g., something that is intentionally doing an array creation to produce a concatenation or to bit select out of the concatenation is totally fine to inline. This is shown in the above example, but concretely producing either of the following is fine (as long as the user intends it):

- concatenation: `assign foo = {{a}, {b}, {c}, {d}};`
- Bit select: `assign bar = {{a}, {b}, {c}, {d}}[idx];`

The one caveat being that this inlining must not produce parentheses.

lattner, 2021-08-17:
Please don't turn on linting as part of check-circt, but maybe in "long tests" mode?  We can't adversely impact developer turnaround cycle.

Ok, it sounds like the requirement is that the array_create is at the top level of the expression.  That should already be achieved by changing your patch to:

```
if (isa<ArrayCreateOp>(op))
      return true;
```

at the right place.

seldridge, 2021-08-17:
> Please don't turn on linting as part of check-circt, but maybe in "long tests" mode? We can't adversely impact developer turnaround cycle.

Let me at least get something together for people to look at. I can bump it to a nightly/long test if needed.

This is a borderline instantaneous check that shouldn't have an effect on developers (it's just checking that the Verilog we're generating conforms to some standard).

This looks like the following invocation and runs in ~100ms on my laptop:

```
# time circt-translate ../test/ExportVerilog/hw-dialect.mlir -export-verilog | verilator --default-language 1800-2017 --bbox-unsup --lint-only /dev/stdin
%Error: /dev/stdin:222:4: syntax error, unexpected ')', expecting '['
  222 |    );
      |    ^
%Error: /dev/stdin:236:4: syntax error, unexpected ')', expecting '['
  236 |    );
      |    ^
%Error: /dev/stdin:336:31: Unsupported: Width of number exceeds implementation limit: 100000'h2CD76FE086B93CE2F768A00B22A00000000000  (IEEE 1800-2017 6.9.1)
  336 |   localparam [99999:0] _tmp = 100000'h2CD76FE086B93CE2F768A00B22A00000000000;  
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: /dev/stdin:423:1: syntax error, unexpected endmodule, expecting ')'
  423 | endmodule
      | ^~~~~~~~~
%Error: /dev/stdin:433:7: syntax error, unexpected IDENTIFIER, expecting ')'
  433 |   San san (  
      |       ^~~
%Error: Exiting due to 5 error(s)
circt-translate ../test/ExportVerilog/hw-dialect.mlir -export-verilog  0.03s user 0.00s system 99% cpu 0.036 total
verilator --default-language 1800-2017 --bbox-unsup --lint-only /dev/stdin  0.05s user 0.01s system 92% cpu 0.072 total
```

Would this be acceptable to have in the developer loop?

teqdruid, 2021-08-17:
I would like this sort of thing to run on each PR/push. I've been toying
around with running the integration tests (with just one of the many
configurations) on each push-to-main. It's short enough (for now) for it
not to be a big deal.

lattner, 2021-08-18:
I have several concerns:
1) We want unit tests to be unit tests, we don't want them to be integration tests.  We have a place for those :-)
2) Not everyone has verilator installed.
3) This is just one sort of lint check, there are dozens of different sorts of things different people may be interested in.
4) Different versions of verilator may have different rules, and we want the testsuite to be as stable and deterministic as possible.

I'd suggest two things: 

1) A couple of things in the integration_tests directory that are largish (e.g. 500 lines, not 50MB :) firrtl files showing some example things, that is specifically about verilator-lint-clean output.
2) A script in circt/utils that can be used on an ad-hoc basis against the whole tree.  It could just take all the .fir / .mlir files in tests (or some specific subdirectories) shove them through firtool and lint it.

The second script can be used now and then to find issues, each of which would then be reduced and tracked as a unit test.

seldridge, 2021-08-18:
I'd be happy with the suggestions you propose. I'd further propose that (1) is not just largish files, but includes unit-tests to give full coverage of `ExportVerilog`. Effectively, just beefing up: https://github.com/llvm/circt/blob/main/integration_test/EmitVerilog/standards.mlir

My high level concern is that Verilog is the foot cannon's foot cannon and it's extremely difficult to both hack on `ExportVerilog` or to review code there without the help of a tool that tells you, "this is legal Verilog." I'm not concerned with being Verilator lint-clean (the compiler will produce correct code), just that we're not generating illegal Verilog.

Your second suggestion of (2) would help periodically sweep for issues as well as give reviewers a tool that they can use to check new Verilog generation behaviors as they are added.

lattner, 2021-08-18:
Makes total sense!

commit 965082c8 (2021-08-16): [ExportVerilog] Don't Inline ArrayGet/ArrayCreate

Block inlining of ArrayCreate into ArrayGet users.  This prevents
ExportVerilog from producing invalid Verilog.  Add one test that locks
in this behavior and update existing tests that incorrectly expected
this.

Fixes #1587.