llvm/circt #1590: [Nightly CI] Integration tests jobs taking >1hr to compile

teqdruid, 2021-08-17:
Example: https://github.com/llvm/circt/runs/3344394208?check_suite_focus=true

Seems like ccache isn't hitting so the unified builds are re-compiling MLIR:
[image]

@mikeurbach Can you take a look? Could be the other configurations are causing cache conflicts? Perhaps it's downloading the wrong ccache?

teqdruid, 2021-08-17:
Seems like it's downloading the wrong ccache. Exactly one job seems to be hitting and taking the amount of time we would expect:

[image]

mikeurbach, 2021-08-17:
Thanks for the investigation, I'll take a look.

mikeurbach, 2021-08-17:
My first hunch is all the different builds in our matrix are actually sharing the same ccache directory. I'm not too sure about how build matrices get split up and isolated, but I'll look into that. If it's not that, I'm not quite sure.

mikeurbach, 2021-08-17:
Another thing I found is the 5GB limit on caches: https://github.com/actions/cache#cache-limits. That appears to be shared between the whole repository, so we may just be evicting caches in some situations. I'll try to see how much cache we can reasonably expect to share and still leave room for the PR cache.

This still doesn't explain situations like this: https://github.com/llvm/circt/runs/3344394208?check_suite_focus=true#step:4:14. We found a cache hit (with the correct/expected cache key), but it still took ~1hr, indicating an LLVM recompile.

I am thinking it is some funny business like I mentioned above.

teqdruid, 2021-08-17:
> This still doesn't explain situations like this: https://github.com/llvm/circt/runs/3344394208?check_suite_focus=true#step:4:14. We found a cache hit (with the correct/expected cache key), but it still took ~1hr, indicating an LLVM recompile.

Look at the cache hit stats output step. It's the wrong cache.

mikeurbach, 2021-08-17:
Yes, but my point is the cache key being restored is set correctly, and the action is downloading the correct cache. I think all of the jobs might be restoring their respective caches into the same location.

teqdruid, 2021-08-17:
> I think all of the jobs might be restoring their respective caches into the same location.

Please elaborate. I don't understand.

I suspect that the ccache action is just ignoring the key (a bug).

mikeurbach, 2021-08-17:
> Please elaborate

We have different caches for each configuration in the build matrix, but each job is restoring them into `$GITHUB_WORKSPACE/.ccache`. I'm not sure how these parallel builds are happening. Are they isolated, so that they each have their own `$GITHUB_WORKSPACE/.ccache`? Or is that directory going to be shared by all of them? If the latter, I can see how that causes this issue--whichever configuration of the build matrix restores its cache last will have a great hit rate, but the others will get screwed. Then again, my gut feeling was each configuration of the build matrix should be isolated and get its own `$GITHUB_WORKSPACE`.

> I suspect that the ccache action is just ignoring the key (a bug).

I've taken a look at the ccache action, it's nothing special. I don't think it's just ignoring the key, it prints out the key it's using. It's just a little TypeScript that [wraps](https://github.com/hendrikmuhs/ccache-action/blob/e22511c559ca6c92f6b13213b41cf00cb51ceaf5/src/restore.ts#L35) the official `cache` action we use on the PR builds.

It only adds a tiny bit of value: a) downloading `ccache` automatically (a feature I actually had to work around--it doesn't work to use `sudo` in our docker container and that was hardcoded), and b) manages a couple configuration values.

I wanted to test my concern that all 16 builds were actually using the same `$GITHUB_WORKSPACE/.ccache` directory, but there's no way to configure that. I've lost all interest in using the `ccache` action at this point, so I am testing just using the officially supported `cache` action. We already have the `ccache` executable in the docker container, so it's easy to switch. I'm testing putting the cache in an explicit, different location for each configuration of the build matrix.

teqdruid, 2021-08-17:
> We have different caches for each configuration in the build matrix, but each job is restoring them into `$GITHUB_WORKSPACE/.ccache`. I'm not sure how these parallel builds are happening. Are they [...]

I assume that each build operates in an isolated environment, though it's possible I'm wrong about that. The downloaded ccache  does get installed to a location which is mapped from the host. Either way, it's very, very unlikely they're sharing a filesystem in the docker hosting VM, so a cache download would just overwrite the last one. 

> I wanted to test my concern that all 16 builds were actually using the same `$GITHUB_WORKSPACE/.ccache` directory, but there's no way to configure that. I've lost all interest in using the `ccache` [...]

That sounds rather more foolproof. Just more maintenance on our end. If this works, we should report a bug to the ccache action authors.

mikeurbach, 2021-08-17:
We'll see how it goes, LLVM is building now over here: https://github.com/llvm/circt/runs/3355053004?check_suite_focus=true. That's from this branch: https://github.com/llvm/circt/pull/1593. Not sure that is exactly right, but I will request review once I get it working.