llvm/circt #1592: [FIRRTL] (unsupported Verilog construct) 2D wires in YoSyS

drom, 2021-08-17:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input arst: AsyncReset
    output fsm_170_next: UInt<4>
    reg fsm_170_state: UInt, clock with: (reset => (arst, UInt(0)))
    when eq(fsm_170_state, UInt<4>(0)) :
      fsm_170_next <= UInt<4>(1)
    else when eq(fsm_170_state, UInt<4>(1)) :
      fsm_170_next <= UInt<4>(2)
    else when eq(fsm_170_state, UInt<4>(2)) :
      fsm_170_next <= UInt<4>(3)
    else when eq(fsm_170_state, UInt<4>(3)) :
      fsm_170_next <= UInt<4>(4)
    else when eq(fsm_170_state, UInt<4>(4)) :
      fsm_170_next <= UInt<4>(5)
    else when eq(fsm_170_state, UInt<4>(5)) :
      fsm_170_next <= UInt<4>(6)
    else when eq(fsm_170_state, UInt<4>(6)) :
      fsm_170_next <= UInt<4>(7)
    else when eq(fsm_170_state, UInt<4>(7)) :
      fsm_170_next <= UInt<4>(8)
    else when eq(fsm_170_state, UInt<4>(8)) :
      fsm_170_next <= UInt<4>(9)
    else when eq(fsm_170_state, UInt<4>(9)) :
      fsm_170_next <= UInt<4>(0)
    else :
      fsm_170_next <= fsm_170_state
    fsm_170_state <= fsm_170_next
```

Compiled with `firtool` produces this Verilog:

```verilog
...
  always @(posedge clock or posedge arst) begin	// a_top_mod.fir:7:5
    if (arst)	// a_top_mod.fir:7:5
      fsm_170_state <= 4'h0;	// a_top_mod.fir:7:5, :8:28
    else	// a_top_mod.fir:7:5
      fsm_170_state <= _T;	// a_top_mod.fir:29:20, :30:19
  end // always @(posedge or posedge)
...
  wire [15:0][3:0] _T_0 = {{fsm_170_state}, {fsm_170_state}, {fsm_170_state}, {fsm_170_state}, {fsm_170_state}, {fsm_170_state}, {4'h0}, {4'h9}, {4'h8}, {4'h7}, {4'h6}, {4'h5}, {4'h4}, {4'h3}, {4'h2}, {4'h1}};	// a_top_mod.fir:8:{10,28}, :9:23, :10:15, :11:23, :12:15, :13:23, :14:15, :15:23, :16:15, :17:23, :18:15, :19:23, :20:15, :21:23, :22:15, :23:23, :24:15, :25:23, :26:15, :29:20
  assign _T = _T_0[fsm_170_state];	// a_top_mod.fir:8:10, :29:20
  assign fsm_170_next = _T;	// a_top_mod.fir:3:10, :29:20
endmodule

```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports this error (wire [15:0][3:0] _T_0 ... line):

```
a_top_mod.fir.mfc.v:60: ERROR: syntax error, unexpected '[', expecting TOK_ID or '#'
```

lattner, 2021-08-18:
@teqdruid any ideas?

seldridge, 2021-08-18:
This may be a great place to start prototyping target triples for CIRCT. 😄

seldridge, 2021-08-19:
Thinking about this a bit more, the most straightforward course of action would be to just convert this pattern to a case statement in `PrettifyVerilog` as had been discussed in the CIRCT meeting a week ago.

lattner, 2021-08-19:
Yes, I agree.

seldridge, 2021-08-20:
There was a discussion yesterday about this and it was proposed that this should _not_ be handled by `PrettifyVerilog`, but by another pass responsible for generating compatible output for what the user wants. Essentially, "prettification" shouldn't be necessary for "correctness" where correctness is defined as what the user says is legal output. The target triples idea probably needs more thought, but this can be prototyped with a new pass that does the cleanup that this issue identifies.

lattner, 2021-08-22:
This should be fixed by 79a17fa1.  Please try passing `-lowering-options=disallowPackedArrays` to firtool when working with Yosys. 

Aliaksei, this is using a bunch of new logic and code paths, could you give this a try and see if Yosys finds it agreeable?

drom, 2021-08-22:
```
module top_mod(
  input        clock, arst,
  output [3:0] fsm_170_next);

  wire [3:0] casez_tmp;	// a_top_mod.fir:28:20
  reg  [3:0] fsm_170_state;	// a_top_mod.fir:6:5

...
  always_comb begin	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
    casez (fsm_170_state)	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0000:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0001:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0010:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0011:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0100:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0101:
        casez_tmp <= fsm_170_state;	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0110:
        casez_tmp <= 4'h0;	// a_top_mod.fir:7:28, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b0111:
        casez_tmp <= 4'h9;	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :24:23, :25:15, :28:20
      4'b1000:
        casez_tmp <= 4'h8;	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :22:23, :23:15, :25:15, :28:20
      4'b1001:
        casez_tmp <= 4'h7;	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :20:23, :21:15, :23:15, :25:15, :28:20
      4'b1010:
        casez_tmp <= 4'h6;	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :17:15, :18:23, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b1011:
        casez_tmp <= 4'h5;	// a_top_mod.fir:9:15, :11:15, :13:15, :15:15, :16:23, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b1100:
        casez_tmp <= 4'h4;	// a_top_mod.fir:9:15, :11:15, :13:15, :14:23, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b1101:
        casez_tmp <= 4'h3;	// a_top_mod.fir:9:15, :11:15, :12:23, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b1110:
        casez_tmp <= 4'h2;	// a_top_mod.fir:9:15, :10:23, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
      4'b1111:
        casez_tmp <= 4'h1;	// a_top_mod.fir:8:23, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
    endcase	// a_top_mod.fir:7:10, :9:15, :11:15, :13:15, :15:15, :17:15, :19:15, :21:15, :23:15, :25:15, :28:20
  end // always_comb
  always @(posedge clock or posedge arst) begin	// a_top_mod.fir:6:5
    if (arst)	// a_top_mod.fir:6:5
      fsm_170_state <= 4'h0;	// a_top_mod.fir:6:5, :7:28
    else	// a_top_mod.fir:6:5
      fsm_170_state <= casez_tmp;	// a_top_mod.fir:28:20, :29:19
  end // always @(posedge or posedge)
  assign fsm_170_next = casez_tmp;	// a_top_mod.fir:2:10, :28:20
endmodule
```

`casez_tmp` has to be typeof `reg`

Or in more details:

```
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:57:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   57 |         casez_tmp <= fsm_170_state;  
      |         ^~~~~~~~~
                    ... For error description see https://verilator.org/warn/PROCASSWIRE?v=4.211
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:59:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   59 |         casez_tmp <= fsm_170_state;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:61:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
[... 33 lines cut ...]
   77 |         casez_tmp <= 4'h6;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:79:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   79 |         casez_tmp <= 4'h5;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:81:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   81 |         casez_tmp <= 4'h4;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:83:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   83 |         casez_tmp <= 4'h3;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:85:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   85 |         casez_tmp <= 4'h2;  
      |         ^~~~~~~~~
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:87:9: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): 'casez_tmp'
                                            : ... In instance top_mod
   87 |         casez_tmp <= 4'h1;  
      |         ^~~~~~~~~
%Error: Exiting due to 16 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

seldridge, 2021-08-23:
I updated this to use `sv.reg` and blocking assignment in c417f11a8d54eb74ff79fc4f3a16efcb50873e90. This gets past any Verilator errors and warnings related to the new code generation.

lattner, 2021-08-23:
Thanks!  Aliaksei, does it work now? :-)

lattner, 2021-08-23:
Curiosity, why does casez_tmp need to be a reg?

seldridge, 2021-08-23:
`wire` is a net type and it can only be driven by an `assign`. `reg` is a variable and can be written by anything. (This isn't making `casez_tmp` a d-flip-flop. It would probably be better to have this be a `logic`, but `reg` works and we already have IR support for it.)

My fix probably isn't ideal, but I'm just trying to get things not broken. (I'm seeing about adding a verifier to disallow `passign` and `bpassign` where the destination is a `sv.wire` to avoid this in the future.)

A more complete fix may be to add types related to `net` and `wire` instead of just `inout` which right now can be either.

drom, 2021-08-23:
Verilog `reg` is just 4-state value type that can be assigned in procedural blocks like always.
It can become, `combinatorial`, `flip-flop` or `latch` depend on `always @?` context.
Verilog `wire` has a `net` type, meant for wiring between modules mostly.

drom, 2021-08-23:
BTW. It is possible to have only one type `reg` and all the logic in `always` blocks,
But impossible to have everything on `wire`

drom, 2021-08-23:
Why do we have `always_comb` ? why not `always @*` ?

drom, 2021-08-23:
the `default` case is missing, so `latch` is inferred.

seldridge, 2021-08-23:
> the default case is missing, so latch is inferred.

`HWLegalizeModules`, the pass that generates these `casez` blocks, won't generate an incomplete case statement. Will that still produce latch inference or will tools do the right thing?

seldridge, 2021-08-24:
@drom: The ordering issue that you saw (case statements were showing up in reverse order) is fixed in e218e1c6fe2f109d4d3ea98d56199887e362c201. 

The `automatic logic` issue (which Yosys also doesn't support) can be worked around by setting a very long line length. When you get a chance, can you add `emittedLineLength` option to your tests? Something like:

```
-lowering-options=disallowPackedArrays,emittedLineLength=8192
```

lattner, 2021-08-24:
It'd be super easy to drop in the last case as a default instead of an explicit value. Lemme know if that helps and I'll happily make it so.

lattner, 2021-08-24:
bdd4c30e ensures that all of the case statements are fully covered with a default.

lattner, 2021-08-24:
Ok, so long as it doesn't make casez_tmp be stateful across invocations of the module then it should be fine I guess.

lemme know more about what we need to do here to improve this.  Are you saying that yosys doesn't support automatic variables at all?  How have we been compatible with it for this long?  I can change the verilog printer to print things as top level declarations if necessary, but I need to know if they should be wires or something else.

lattner, 2021-08-24:
I filed the "automatic logic" issue in [this ticket](https://github.com/llvm/circt/issues/1633).

seldridge, 2023-11-21:
I'm pretty sure this is fixed with the current behavior of `-lowering-options=disallowPackedArrays`. Recent work like https://github.com/llvm/circt/pull/6402 gets bigger projects like Chipyard working with Yosys.

closed 2023-11-21 by seldridge