llvm/circt #1594: [LowerToHW] Register w/ Wire Reset Produces Two Always Blocks

seldridge, 2021-08-17:
As identified https://llvm.discourse.group/t/firtool-register-verilog-emitter/4097, formulations of registers that are reset by wires may cause multiple always blocks to be generated. 

Consider:

```mlir
module  {
  firrtl.circuit "MyTop"   {
    firrtl.module @MyTop(in %clk: !firrtl.clock, in %reset: !firrtl.uint<1>, in %a: !firrtl.uint<1>, out %b: !firrtl.uint<2>) {
      %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
      %reset_n = firrtl.wire  : !firrtl.uint<1>
      %0 = firrtl.not %reset : (!firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.connect %reset_n, %0 : !firrtl.uint<1>, !firrtl.uint<1>
      %r = firrtl.regreset %clk, %reset_n, %c0_ui1  : !firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>
      firrtl.connect %r, %a : !firrtl.uint<1>, !firrtl.uint<1>
      firrtl.connect %b, %r : !firrtl.uint<2>, !firrtl.uint<1>
    }
  }
}
```

This is getting lowered to two always blocks:

```mlir
  hw.module @MyTop(%clk: i1, %reset: i1, %a: i1) -> (%b: i2) {
    %true = hw.constant true
    %false = hw.constant false
    %reset_n = sv.wire  : !hw.inout<i1>
    %0 = comb.xor %reset, %true : i1
    sv.assign %reset_n, %0 : i1
    %1 = sv.read_inout %reset_n : !hw.inout<i1>
    %r = sv.reg  : !hw.inout<i1>
    sv.alwaysff(posedge %clk)  {
    }(syncreset : posedge %1)  {
      sv.passign %r, %false : i1
    }
    %2 = sv.read_inout %reset_n : !hw.inout<i1>
    sv.alwaysff(posedge %clk)  {
      sv.passign %r, %a : i1
    }(syncreset : posedge %2)  {
    }
    %3 = sv.read_inout %r : !hw.inout<i1>
    %4 = comb.concat %false, %3 : (i1, i1) -> i2
    hw.output %4 : i2
  }
```

This should produce only one always block.

seldridge, 2021-08-17:
Okay, this is sillier than I thought. The problem is that the resets look different because CSE hasn't run. The reset of the first block is `%1` and the reset of the second block is `%2`, but they're both the same read of `%reset_n`. Moving CSE before HWCleanup fixes this.

lattner, 2021-08-18:
Thank you for tracking down the root issue here @seldridge, I think this patch is the right way to go.  LowerToHW already auto-cse's Always blocks, so we want to make sure it is doing this for readinout.  This also produces easier to read IR, which is nice.  :-)

Fixed in b7c86896

closed 2021-08-18 by lattner

seldridge, 2021-08-18:
Your fix here looks good to me. Thanks!