llvm/circt #1597: [Calyx] Define a Cell Interface.

cgyurgyik, 2021-08-18:
A `Cell` compromises of Calyx component instances and primitives.

An `InstanceOp` retrieves its port names and directions from the referenced component. A primitive should define port names and directions. To guarantee this, we can use an interface such as the one below:

```
/// Op Interface for cells.
def CellInterface : OpInterface<"CellInterface"> {
  let cppNamespace = "::circt::calyx";
  let description = [{
    This is an op interface for Calyx Cells. Cells consist
    of primitives and instances of components.
  }];
  let methods = [
    InterfaceMethod<
      "This returns the port names associated with the cell.",
      "SmallVector<StringRef>", "portNames"
    >
    // Also add interface method for port directions.
  ];
}
```

We can then add `DeclareOpInterfaceMethods<CellInterface>` to the `CalyxCell` class to ensure each `Cell` has the proper methods defined.

mikeurbach, 2021-08-24:
I can work on this.

mikeurbach, 2021-09-01:
I introduced the basic interface here: https://github.com/llvm/circt/pull/1695. In that PR, I haven't dealt with the TODOs above yet.

mikeurbach, 2021-09-01:
I landed the interface, so we can start using it. If anyone else wants to loop back to the TODOs in this issue, feel free, or else I'll tackle them at some point.

cgyurgyik, 2021-09-02:
I'll take care of the AssignOp verification.

cgyurgyik, 2021-09-10:
Completed by @mikeurbach in #1695.

closed 2021-09-10 by cgyurgyik