llvm/circt #1600: test/ExportVerilog/sv-dialect.mlir: issue439 testcase generates invalid verilog

lattner, 2021-08-18:
Sez @seldridge in [this comment](https://github.com/llvm/circt/commit/e043814d2c1fcf418070a5f977c96ba75451bee5): 

I've been taking a look at sv-dialect.mlir to make sure that this is generating legal Verilog. The test added here is actually invalid Verilog. The full output for the added test is:

```
module issue439(
  input in1, in2);

  wire _T;      // issue439.mlir:7:3
  wire _T_0;    // issue439.mlir:10:15

  assign _T = in1 | in2;        // issue439.mlir:4:12, :7:3
  always @(posedge _T) begin    // issue439.mlir:7:3
    _T_0 <= in1;        // issue439.mlir:10:15
    _T_0 <= in2;        // issue439.mlir:10:15
    $fwrite(32'h80000002, "Bye %x\n", _T_0);    // issue439.mlir:10:15, :12:5
  end // always @(posedge)
endmodule
```

Problematically, wire can only be driven from an assign (any blocking or blocking assignment, = or <=, is illegal for wires).

I'm not entirely sure what comb.merge is supposed to be modeling, but I think this really wants to be a SystemVerilog inout and either connected by alias or a bi-directional connect. (Tools will accept different things here.) This is the lowering that FIRRTL analog types are using and is probably fine to re-use here.

lattner, 2021-08-18:
Great catch.  "comb.merge" is attempting to model an SSA-based answer to "this wire is being driven by multiple things and only one of them is non-z at a time".  That said, this is not actually pulling its weight in the compiler.  I'll just remove it entirely.

lattner, 2021-08-18:
Ding dong, comb.merge is dead: f0fa80b4

closed 2021-08-18 by lattner

seldridge, 2021-08-18:
🏠 🧦  Thanks for removing this! This seems like the right fix for now.