llvm/circt #1624: [Comb] `or` of exclusive concats can be merged

lattner, 2021-08-22:
I'm seeing a bunch of variants of this pattern:

// x is 6 bits, y is 2 bits.
`wire [9:0] _tmp_122 = {3'h0, x} | {y, 7'h0};`

which could be simplified to:

`wire [9:0] _tmp_122 = {y, 1'h0, x};`

A similar one is:

`{ 5'0, bit}|{4'0, cond?2:0}` ==> `{4'h0, cond ? 1 : 0, bit}` ==> `{4'h0, cond, bit}`

lattner, 2021-09-11:
Random thought - the second pattern can be handled by canonicalizing `cond?2:0` -> `concat(cond, 0)` which would then cause it to compose into the existing transformation.

youngar, 2024-12-23:
https://github.com/llvm/circt/pull/1701 is causing performance issues and I am going to revert it until the problems can be addressed.

mikeurbach, 2024-12-23:
Funny enough we once hit a performance cliff in verilator for a similar situation https://github.com/verilator/verilator/issues/3405. I think the fix there ended up being some kind of "optimization fuel" like approach where it just stopped if the expression was large... That could maybe work, but we should probably come up with something that's both general and performant for ourselves.

commit a69675b7 (2021-09-26): [Comb] Canonicalize mux with constant inputs (#1820)

This handles several cases of muxes with constant input values:

- mux(a, 0, b) -> and(~a, b), for single-bit b.

- mux(a, b, 1) -> or(~a, b), for single-bit b.

- mux(a, b, c) -> concat(d, a, e), when b and c are constants that differ by
  only one bit.

  A common special case of this is when b is a power of 2 and c is 0, where
  expression can be simplified by "pushing" the mux condition a into the value
  of b.

  E.g. mux(a, 2, 0) -> concat(a, 0).

Fixes #1624.