llvm/circt #1653: [LLHD] Add lowering from `llhd.eq` to `comb.eq`

fabianschuiki, 2021-08-28:
LLHD has a `llhd.eq` and `llhd.neq` operation which works on a lot of different types, including things like structs and arrays. Add a lowering pass that maps these operations to the corresponding `comb.eq` (or similar) ops, possibly creating multiple such ops that individually compare struct fields or vector indices.

maerhart, 2021-09-05:
As we are using the HW types we could also just insert a `hw.bitcast` for each operand to cast any array or struct (possibly containing other arrays or structs) to a big integer and just use one `comb.icmp eq`. This reduces the size of the IR (we would otherwise need field/element accesses and a compare for each element) and bitcast has no cost in hardware and also shouldn't be a problem in simulation. Of course, this only works when all 'leaf' types are integers.

Which types are allowed inside arrays and structs? The LLHD reference does not mention any restriction and I've seen integers and time in the example there, so are pointers and signals also allowed? I'm also wondering how comparison of time should be lowered. Theoretically, it's just a struct of three integers, but we don't have any instruction to perform a cast on the LLHD/Comb level.

fabianschuiki, 2021-09-06:
Hmmm yeah a bitcast sounds like a good idea. We probably only want to allow integer types in the leafs, as you suggest. We'll need time comparison/arithmetic at some point, but it'll probably be easier to introduce separate operations for that as soon as Moore emits them. I expect these to mainly originate in SV time arithmetic, like `time t1; time t2; #(t1+t2) $display("Hello");`, to compute `t1+t2`.

maerhart, 2021-09-06:
Ok sounds great! Thanks for taking the time to think about this.

Considering we only allow integers in aggregates, I'm wondering whether it is even worth having those two instructions in LLHD.
They work on integers, arrays, structs, and pointers. As `llhd.eq` compares the stored value of pointers and not memory addresses we only need one `llhd.load` per operand and then do the same as for the other types. This means, these two ops replace 2 loads, 2 bitcasts, and 1 compare in the worst case. I think the IR would be more 'canonical' when using explicit loads and thus not support pointers here.

I think it could be worth having these ops if comparison of aggregates is very common as it would then decrease the IR size, but I haven't really seen moore emit comparisons of aggregates so far. Is it worth having two ops plus a lowering pass to spare the frontend emitting two bitcasts per compare of aggregate types?

I'd be interested to hear what advantages and disadvantages you see in having these two ops?

I agree that for the time comparison it would then probably be best to mirror `comb.icmp` for time values only.

fabianschuiki, 2021-09-06:
That's an excellent point! I fully agree that it's probably better to err on a simpler, more "canonical" IR first and get that to work. And if we ever see that there's a frequent need to encode aggregate comparisons (which I highly doubt, now that I'm thinking of it), we can always add them. Generating the code to do comparisons is definitely something a SV/VHDL frontend can do!

I'd also be happy with just saying "we don't support time arithmetic at the moment", and then come back to that when Moore can actually emit such code.

In the meantime I'm trying to slowly nudge Moore into the MLIR domain, make it link against CIRCT, and use its C API to emit code. Then we can try to gradually raise the handover point between the custom Rust IRs and CIRCT. First that'll be LLHD/Comb/Seq/HW, but later I'd love to try and migrate the Moore *MIR* layer over to MLIR.