llvm/circt #1656: [LLHD] Replace inserts and non-signal extracts with Comb ops

fabianschuiki, 2021-08-28:
See how far we can replace the LLHD insert and extract operations, which modify structs, arrays, and bits, with operations in the Comb dialect. Comb might call these ops quite differently, so this needs some investigation. Also, some of the LLHD extract ops work on signals, so in a sense are lvalue variants, which likely won't have an equivalent in Comb. We might want to make the LLHD signal ops just be lvalue-flavored duplicates of the Comb operations, to better align with that dialect.

maerhart, 2021-09-01:
I came up with the following mapping. Does that roughly match with what you have in mind?

#### `llhd.insert_slice`

Array:
```mlir
llhd.insert_slice %target, %slice, 4 : !hw.array<32xi32>, !hw.array<16xi32>
```
would become
```mlir
%c20 = hw.constant 20 : i5
%c0 = hw.constant 0 : i5
%a = hw.array_slice %target at %c20 : (!hw.array<32xi32>) -> !hw.array<12xi16>
%b = hw.array_slice %target at %c0 : (!hw.array<32xi32>) -> !hw.array<4xi16>
hw.concat %a, %slice, %b : (12, 16, 4 x i32)
```

Integer: same but with `comb.extract` and `comb.concat`

#### `llhd.insert_element`

Struct: `hw.struct_inject`
Array: same as insert slice, but requires to create an array with one value for the slice using `hw.array_create`

#### `llhd.extract_slice`

Integers: `comb.extract`
Arrays: use dynamic extract
Signal of integer: use dynamic extract
Signal of array: use dynamic extract

#### `llhd.dyn_extract_slice`

Integers: shift + `comb.extract`
Arrays: `hw.array_slice`
Signal of integer: custom op mirroring `comb.extract` but with dynamic start-index
Signal of array: custom op mirroring `hw.array_slice`

#### `llhd.extract_element`

Array: use dynamic extract
Struct: `hw.struct_extract`
Signal of Array: use dynamic extract
Signal of Struct: custom op mirroring `hw.struct_extract`

#### `llhd.dyn_extract_element`

Array: `hw.array_get`
Signal of Array: custom op mirroring `hw.array_get`

This means we replace the current 2 insert and 4 extract operations with 4 extract operations that work on one type each.
Additionally we have to implement (or modify) conversion patterns to LLVM for `hw.array_slice`, `hw.concat`, `hw.array_get`, `hw.struct_inject`, and `hw.struct_extract` to maintain the same `llhd-sim` functionality.

Any naming suggestions for the new operations?

fabianschuiki, 2021-09-02:
This looks great! Thanks a lot for tackling this. Regarding naming, you could try to mirror the HW names as you suggest, but maybe prefix them with `llhd.sig.`, so it's clear that they belong into the "signal namespace". Like `llhd.sig.struct_extract`.

commit 3453997c (2021-10-30): [LLHD] Delete insert ops, revise extract ops (#1992)

Closes #1656 by removing the LLHD insert operations and replacing the extract operations as described in the issue.

Some notes/observations:
- Semantics of accessing out of bounds for both slices and single elements in LLHD were partially implemented to return a default value depending on the type (only for non-signal arrays in `llhd.dyn_extract_slice`) or UB. This PR defines it to be UB as this is also what HW does (I think).
- Field accesses to structs are checked, though.
- Extract slice operations cannot return a wider type (at most eq [... cut]