llvm/circt #1658: [LLHD] Replace `llhd.const` with `hw.constant`

fabianschuiki, 2021-08-28:
LLHD uses its own `llhd.const` operation to materialize integer and time constants. The integer part should be replaced with `hw.constant` so we can benefit from the infrastructure being established in HW/Comb. The remaining LLHD op could then become a dedicated time constant op, like `llhd.constant_time 1s, 3d, 9e`.

maerhart, 2021-08-28:
I'm wondering, what's the reason that the constant operation is located in the `HW` dialect instead of the `comb` dialect? I haven't really followed the development of those dialects, but to me it seems the `HW` dialect depends on the operations in `comb` and `comb` materializes to the `hw.constant` operation. Also, I had to include both dialects in the lowering to LLVM, because without adding a conversion pattern for the `hw.constant` it failed to legalize. So wouldn't it make more sense to have `comb.constant` (as comb aims to be the more generic dialect used by other dialects)? Or am I missing something?

But I agree, sharing the same constant operation is a good idea.

fabianschuiki, 2021-08-28:
I think the idea was to have HW as an "umbrella" dialect that captures common types and structures such as modules and extmodules. Comb then adds all the non-controversial operations that don't interact with timing/state, based on the types in HW. That's also why Comb ops generally live in `hw.module` and operate on `hw.constant`s.

The intended pattern if I recall correctly is that you always depend on HW, and then pull in the other stuff you need, like Comb or Seq. If HW needed Comb to link properly, that *might* be a design smell, or it has been a conscious design decision in the past :thinking:.

lattner, 2021-08-28:
Yeah, we want a clean dependence graph here.  comb/seq/sv are layered on top of HW.  SV depends on comb in some places.  HW is where the completely generic stuff goes and was found to be the best fit for constant for reasons.

maerhart, 2021-08-29:
Thank you for explaining! I also found the related issues and PRs. Makes sense to me now.

PR #1702 (2021-09-03): [LLHD] Replace llhd.const with hw.constant
Closes #1658
closed 2021-09-03 by maerhart