llvm/circt #1666: [Calyx] Add attributes to Calyx ports.

cgyurgyik, 2021-08-30:
When emitting Calyx, interface ports like `go`, `done`, `reset` and `clk` must be marked using attributes like `@go`, `@done`, `@clk`, and `@reset`. Otherwise the Calyx frontend will automatically add new ports with those attributes attached to them.

You'd want to generate something like:
```
component A(in: 8, @go go: 1, @clk clk: 1, @reset reset: 1) -> (out: 8, @done done: 1) {
```

_Originally posted by @rachitnigam in https://github.com/llvm/circt/issues/1662#issuecomment-908079077_

For the emitter, we can simply hard-code these for components. I think it would be nice to create and attach these custom attributes within the Calyx Dialect as well. Given that a `ComponentOp` lists all Inputs/Outputs as arguments, I think we can attach these attributes directly to the arguments.

mortbopet, 2021-09-04:
Changes should also be made to the filtering of interface ports function in the cell interface:
https://github.com/llvm/circt/blob/44229b44cdd8d5b96a2f097fa0cc69d0625faa32/include/circt/Dialect/Calyx/CalyxInterfaces.td#L112-L124

cgyurgyik, 2021-09-05:
Going to take a crack at this.

rachitnigam, 2021-09-05:
This test file shows almost all places where attributes can be placed: https://github.com/cucapra/calyx/blob/master/tests/parsing/attributes.expect

rachitnigam, 2021-09-05:
Once this is done, we should add an issue in the Calyx native repo to emit attributes in the CIRCT backend

closed 2021-09-08 by cgyurgyik