llvm/circt #1677: [SV]/[HW] IR Support for Comments

seldridge, 2021-08-31:
Add IR support for representing comments, including block comments.  `sv::VerbatimOp` is extremely useful here, but it forces the pass that generates them to reason about things like line length, how to handle newlines, whether it should generate line feeds, carriage returns, or something else etc.  It would be more ergonomic to have an `sv::CommentOp` or `sv::BlockCommentOp` which a pass can just dump a string into and have these show up correctly during emission.

lattner, 2021-08-31:
Makes sense.  This could be statement level (an sv.comment op) or module level (an attribute on the module).

seldridge, 2021-08-31:
Supporting both ops and attributes sounds great. I was worried about how to "anchor" comments to statements. (It seems totally overkill to have a comment op variant that takes a block (and also that would cause further symbol table issues).)

lattner, 2021-08-31:
A more stable thing would be to add attribute support for other declarations like wire/reg etc and control flow statements like `sv.if`, so they don't wander.

teqdruid, 2021-09-03:
A related must-have feature for me is annotation comments which are attached to specific statements. Example:

```
(* altera_attribute = {"-name DUPLICATE_REGISTER 5"} *)
reg data_in;
```

seldridge, 2021-10-31:
Supporting both comment attributes (in the HW dialect) and _Verilog Attribute_[^1] attributes (in the SV dialect) seems straightforward. (I'm taking a look at this as this would simplify some confusion that arises from using `sv::VerbatimOp` to encode comments coming from a FIRRTL pass.)

One question on implementation. I see several ways of doing the actual emission from `ExportVerilog`:

1. Teach all operation visitors about the comment/_Attribute_ attribute so that these get emitted.
2. Use a templated function to save on the boilerplate of (1).
3. Do a pre-pass that expands all comments/attributes into actual `sv::VerbatimOp` before lowering.

Any thoughts on a DRY way of handling this in `ExportVerilog`?

[^1]: It is annoying, but these in-line annotation comments with `(* ... *)` are called "Attributes" in Verilog.

lattner, 2021-10-31:
I think that `//` and `/*` comments in verilog statements should be handled fine with verbatim blocks.  I can see utility in having a first-class way to attach "decl comments" and "Verilog attributes" explictly on sv.reg, hw.module, etc.

I don't think the later needs to substitute symbol names and expressions in - or at least I wouldn't do that proactively unless you know there is a need or it already.