llvm/circt #1681: [FIRRTL] (>>> ^) mismatch

drom, 2021-08-31:
The following FIRRTL program

```python
; seed: 7564
circuit top_mod :
  module top_mod :
    input inp_db: SInt<18>
    output _tmp53: UInt<18>
    wire tmp49: SInt<18>
    tmp49 <= dshr(inp_db, tail(asUInt(inp_db), 11))
    _tmp53 <= xor(tmp49, asSInt(UInt<13>(2856)))
```

Compiled with `firtool   --lower-to-hw --infer-widths --imconstprop --lowering-options=disallowPackedArrays,emittedLineLength=8192 --mlir-timing --verilog -o=$VFILE2` produces this Verilog:

```verilog
module top_mod(	// a_top_mod.fir:3:10
  input  [17:0] inp_db,
  output [17:0] _tmp53);

  assign _tmp53 = $signed(inp_db) >>> $signed(inp_db[6:0]) ^ 18'hB28;	// a_top_mod.fir:3:10, :7:{14,27}, :8:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [17:0] inp_db,
  output [17:0] _tmp53
);
  wire [17:0] _GEN_0 = inp_db;
  wire [17:0] tmp49 = $signed(inp_db) >>> _GEN_0[6:0];
  assign _tmp53 = $signed(tmp49) ^ 18'shb28;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 18 unproven $equiv cells in 'equiv_status -assert'.
```

drom, 2021-08-31:
Result of `$signed(inp_db) >>> $signed(inp_db[6:0])` is unsigned.

Instead of: 
```
assign _tmp53 = $signed(inp_db) >>> $signed(inp_db[6:0]) ^ 18'hB28;
```

MFC should be doing:

```
assign _tmp53 = $signed($signed(inp_db) >>> $signed(inp_db[6:0])) ^ 18'hB28;
```

lattner, 2021-08-31:
I'm not exactly sure what is going on here, but I don't think it has to do with the signedness.  Does Yosys provide an example input that demonstrates the problem.   I suspect it has to do with the explicit conversion of the shift result to 18 bits.

lattner, 2021-08-31:
actually, it is a precedence issue.  We shuld be parenthesizing the LHS of the ^

lattner, 2021-08-31:
actually, no, it isn't a precedence issue.  >>> bind tighter than ^

fabianschuiki, 2021-09-01:
The problem could be in the RHS of the shift. MFC is truncating `inp_db` to 7 bits, then sign-extending that back to 18 bits (implicitly because it's an operand to `>>>` of 18 bit size), while SFC zero-extends the 7 bits to 18 bits. I'm not sure anymore how exactly SV handles signed shift amounts, but AFAIK they are just treated as unsigned numbers. So if you pass in a `0x40`, SFC would right-shift by 64, and MFC would right-shift by 262112 (`0x3FFE0`).

lattner, 2021-09-01:
drom is right, changing it from ` ^ 18'hB28;` to ` ^ 18'shB28;` makes it work according to icarus.

lattner, 2021-09-01:
Ok, after reading the spec, I still really have no idea how this works, but it is clear that the signedness of the `>>>` operator is determined from its context as well as its operand, as shown in this example:

```
module tb;

  reg [15:0] a;
  wire[15:0] b;
  wire[15:0] c;

  top_mod Instance (.inp_db(a), ._tmp53(b));
  top_mod2 Instance3 (.inp_db(a), ._tmp53(c));
    
    initial
        begin
         a = 16'hF002;
         $fwrite(32'h80000002, "%x %x\n",
                 ($signed(a) >>> 4) ^ 16'sh0,
                 ($signed(a) >>> 4) ^ 16'h0);
         
          # 10
         $fwrite(32'h80000002, "%x %x %x\n", b, c, ($signed(a) >>> 4) ^ 16'h0);
          
          
        end
endmodule

module top_mod(
  input  [15:0] inp_db,
  output [15:0] _tmp53
);
  wire [15:0] _GEN_0 = inp_db;
  wire [15:0] tmp49 = $signed(inp_db) >>> _GEN_0[6:0];
  assign _tmp53 = $signed(tmp49) ^ 16'sh1;
endmodule

module top_mod2(
  input  [15:0] inp_db,
  output [15:0] _tmp53);

  assign _tmp53 = ($signed(inp_db) >>> 2) ^ 16'h1;	
endmodule
```

I will just change >>> printing to wrap the result with a $signed.  That will ensure we get the right thing.  So nasty.

lattner, 2021-09-01:
> The problem could be in the RHS of the shift. MFC is truncating inp_db to 7 bits, then sign-extending that back to 18 bits (implicitly because it's an operand to >>> of 18 bit size), while SFC zero-extends the 7 bits to 18 bits. 

Actually, the RHS of shift operands works different than other binary operators in verilog.  They are "self determined" which means they don't participate in sign or width calculation.  They are always treated as unsigned values.

lattner, 2021-09-01:
Fixed in 31070d0e, thank you so much for identifying this nasty issue @drom !

closed 2021-09-01 by lattner