llvm/circt #1690: [Calyx] Round tripping with the native Calyx compiler

rachitnigam, 2021-09-01:
A short term objective identified in the August 31 Calyx + CIRCT meeting is enabling round tripping between the Calyx CIRCT and Calyx native representations. This issue tracks progress in both the Calyx native and CIRCT repositories for the issues that need to be fixed.

## MVP
- [ ] https://github.com/llvm/circt/issues/1679
- [ ] Representation for Calyx primitives in CIRCT (@mortbopet had thoughts on this)
- [ ] https://github.com/cucapra/calyx/issues/642
- [ ] https://github.com/llvm/circt/issues/1666

We can consider this "done" when the Calyx files in the native test suite can be round tripped.

(I don't have permissions to add the right labels to issue so apologies for that)

rachitnigam, 2021-09-04:
The Calyx native compiler's implementation for combinational groups is going to land soon which will be a breaking change: https://github.com/cucapra/calyx/pull/635

The important changes are:
1. Introduce `comb group` syntax which defines a combinational group. A combinational group does not have a done condition and cannot be used in regular control operators.
2. Make the `with <group>` syntax optional for `if` and `with`. Additionally, only combinational groups are allowed in `with` positions

The compilation process will also change to:
1. Use the [RemoveCombGroups](https://capra.cs.cornell.edu/docs/calyx/source/calyx/passes/struct.RemoveCombGroups.html) pass to transform `comb group` to `group` and hoist `with <comb_group>` from `if` and `while` control operators.
2. The FSM generation pass will require that `if` and `with` do not have anything in the `with` position.

Happy to have a longer conversation about this to align the syntax & semantics.

rachitnigam, 2021-09-05:
Okay folks, we're almost there! Once #1723 & #1724 are fixed, we should be able to completely roundtrip the emit.mlir example.

mortbopet, 2021-09-05:
> Make the with <group> syntax optional for if and with. Additionally, only combinational groups are allowed in with positions

Good to know - this will simplify #1630 a bit.
Should we have a check for asserting that the `if/when $cond ` cond value is driven by the preceding combinational group? This is already done for `if $cond with $group`, i.e. `group` must drive `cond`.

cgyurgyik, 2021-09-05:
Removed #1666; I don't think is necessary for the completion of a basic emitter.

rachitnigam, 2021-09-06:
> Should we have a check for asserting that the if/when $cond cond value is driven by the preceding combinational group? This is already done for if $cond with $group, i.e. group must drive cond.

@mortbopet, The difference between `if <comb_port> with <comb_group>` and `if <seq_port>` is that in the former case, the Calyx program needs to run `<comb_group>` to ensure that `<comb_port>` has a valid value. On the other hand, in the latter case, the program must ensure that the value in `<seq_port>` is relevant when the `if` executes.

This means that in the `<seq_port>` case, the port is going to be connected to a sequential element (like a register) while in the `<comb_port>`, it's likely to be a combinational.

For example:
```
seq {
  one;
  two;
  if reg.out { ... }
}
```
We know two things here: `one` and `two` are not combinational groups (since combinational groups are not allowed in normal control flow) & `reg.out` is a port on a sequential port. The program is free to modify it in `one` and not touch it in `two`.

Does this make sense?

rachitnigam, 2021-09-12:
If @mortbopet & @cgyurgyik are in agreement, let's close this tracker PR in favor of a new one that tracks issues that need to be fixed to get #1630 working. As things stand, we can roundtrip the Calyx dialect tests already. We can come back to implementing `invoke` when we need and instead focus on getting SCF programs simulating through Verilator using a calyx native flow.

cgyurgyik, 2021-09-12:
SGTM

mortbopet, 2021-09-12:
Good with me!

closed 2021-09-13 by rachitnigam