llvm/circt #1700: Performance Regression (After LLVM Merge?)

seldridge, 2021-09-02:
There's a pretty comical performance regression that showed up last night. This is suspected to be related to the LLVM merge.

[image]

See: [circt.org/perf](https://circt.org/perf)

lattner, 2021-09-03:
Filed here: https://bugs.llvm.org/show_bug.cgi?id=51738

lattner, 2021-09-08:
Patch here:
https://reviews.llvm.org/D109454

lattner, 2021-09-08:
Landed in https://reviews.llvm.org/D109454  .  Could someone merge in mainline LLVM to pick this up?

seldridge, 2021-09-08:
Yeah, I'm on it... Thanks for the fix, @lattner!

closed 2021-09-09 by seldridge

seldridge, 2021-09-09:
This _appears_ to be fixed for me locally. I'm kicking off the process of getting circt.org/perf updated, but that takes a while.  Locally, using a debug build and `-verify-each=false`, I'm getting the following results for `test2` which don't look heinously broken:

```
===-------------------------------------------------------------------------===
                         ... Execution time report ...
===-------------------------------------------------------------------------===
  Total Execution Time: 162.1973 seconds

  ----Wall Time----  ----Name----
   33.9496 ( 20.9%)  FIR Parser
  108.4516 ( 66.9%)  'firrtl.circuit' Pipeline
    8.9351 (  5.5%)    'firrtl.module' Pipeline
    5.1524 (  3.2%)      CSE
    0.0000 (  0.0%)        (A) DominanceInfo
    3.7827 (  2.3%)      LowerCHIRRTL
    0.9324 (  0.6%)    InferWidths
    1.7590 (  1.1%)    InferResets
    0.1164 (  0.1%)      (A) circt::firrtl::InstanceGraph
    3.9889 (  2.5%)    LowerFIRRTLTypes
   56.4167 ( 34.8%)    'firrtl.module' Pipeline
    5.9567 (  3.7%)      ExpandWhens
   50.4600 ( 31.1%)      Canonicalizer
    0.0657 (  0.0%)    Inliner
    5.6653 (  3.5%)    IMConstProp
    0.0917 (  0.1%)      (A) circt::firrtl::InstanceGraph
    0.0000 (  0.0%)    BlackBoxReader
   30.6879 ( 18.9%)    'firrtl.module' Pipeline
   30.6879 ( 18.9%)      Canonicalizer
   19.7794 ( 12.2%)  Print .mlir output
    0.0161 (  0.0%)  Rest
  162.1973 (100.0%)  Total
```

lattner, 2021-09-09:
Thank you @seldridge !  Yes, this looks inline with expectations.  @drom - when we get verification that the graphs are normal tomorrow, can you please manually expunge the bad datapoints from the graph (otherwise the zoom level is not useful)

commit 49b84146 (2021-09-08): Bump LLVM for Canonicalizer Perf. Regression Fix

Fixes #1700.
PR #1750 (2021-09-09): Bump LLVM for Canonicalizer Perf. Regression Fix
Fixes #1700.
commit 50ddd922 (2021-09-09): Bump LLVM for Canonicalizer Perf. Regression Fix (#1750)

Bump LLVM to fix a performance regression in the canonicalizer.  This
introduced an ordering change in one canonicalizer test.

Fixes #1700.