llvm/circt #1705: [FIRRTL] (mem) Procedural assignment to wire

drom, 2021-09-03:
The following FIRRTL program

```python
; seed: 1
circuit top_mod :
  module top_mod :
    input clock: Clock
    input inp_bb: SInt<24>
    input inp_g: UInt<0>
    mem tmp173:
      data-type => SInt<5>
      depth => 2
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
      reader => r0
      writer => w0
    tmp173.r0.clk <= clock
    tmp173.r0.en <= head(UInt<4>("h6"), 1)
    tmp173.r0.addr <= asUInt(SInt<30>("o-744515272"))
    tmp173.w0.clk <= clock
    tmp173.w0.en <= head(asUInt(inp_bb), 1)
    tmp173.w0.addr <= inp_g
    tmp173.w0.data <= asSInt(inp_g)
    tmp173.w0.mask <= inp_g
```

Compiled with `firtool   --lower-to-hw --infer-widths --imconstprop --lowering-options=disallowPackedArrays,emittedLineLength=8192 --mlir-timing --verilog -o=$VFILE2` produces this Verilog:

```verilog
module FIRRTLMem_1_1_0_5_2_0_1_0(	// a_top_mod.fir:7:5
  input        ro_clock_0, ro_en_0, ro_addr_0, wo_clock_0, wo_en_0, wo_addr_0, wo_mask_0,
  input  [4:0] wo_data_0,
  output [4:0] ro_data_0);

  reg [4:0] Memory[0:1];

  wire _T = wo_en_0 & wo_mask_0;
  wire [4:0] _T_0 = Memory[wo_addr_0];
  always @(posedge wo_clock_0) begin
    if (_T)
      _T_0 <= wo_data_0;
  end // always @(posedge)
  assign ro_data_0 = ro_en_0 ? Memory[ro_addr_0] : 5'bx;	// a_top_mod.fir:7:5
endmodule

module top_mod(	// a_top_mod.fir:3:10
  input        clock,
  input [23:0] inp_bb);

  wire [4:0] tmp173_ro_data_0;	// a_top_mod.fir:7:5

  FIRRTLMem_1_1_0_5_2_0_1_0 tmp173 (	// a_top_mod.fir:7:5
    .ro_clock_0 (clock),
    .ro_en_0    (1'h0),	// a_top_mod.fir:16:21
    .ro_addr_0  (1'h0),	// a_top_mod.fir:16:21
    .wo_clock_0 (clock),
    .wo_en_0    (inp_bb[23]),	// a_top_mod.fir:19:21
    .wo_addr_0  (1'h0),	// a_top_mod.fir:16:21
    .wo_mask_0  (1'h0),	// a_top_mod.fir:16:21
    .wo_data_0  (5'h0),	// a_top_mod.fir:21:20
    .ro_data_0  (tmp173_ro_data_0)
  );
endmodule
```

```
Verilator 4.211 devel rev v4.210-70-gc69ddc46
```

Gives the following error:

```
%Error-PROCASSWIRE: a_top_mod.fir.mfc.v:12:7: Procedural assignment to wire, perhaps intended var (IEEE 1800-2017 6.5): '_T_0'
                                            : ... In instance top_mod.tmp173
   12 |       _T_0 <= wo_data_0;
      |       ^~~~
                    ... For error description see https://verilator.org/warn/PROCASSWIRE?v=4.211
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

seldridge, 2021-09-07:
This is likely a bug with ExportVerilog incorrectly treating `_T_0` like a wire when it's actually a reference to a memory location.

darthscsi, 2021-09-08:
mlir is correct:
```  hw.module @FIRRTLMem_1_1_0_5_2_0_1_0(%ro_clock_0: i1, %ro_en_0: i1, %ro_addr_0: i1, %wo_clock_0: i1, %wo_en_0: i1, %wo_addr_0: i1, %wo_mask_0: i1, %wo_data_0: i5) -> (%ro_data_0: i5) {
    %Memory = sv.reg  : !hw.inout<uarray<2xi5>>
    %0 = sv.array_index_inout %Memory[%ro_addr_0] : !hw.inout<uarray<2xi5>>, i1
    %1 = sv.read_inout %0 : !hw.inout<i5>
    %x_i5 = sv.constantX : i5
    %2 = comb.mux %ro_en_0, %1, %x_i5 : i5
    sv.alwaysff(posedge %wo_clock_0)  {
      %3 = comb.and %wo_en_0, %wo_mask_0 : i1
      sv.if %3  {
        %4 = sv.array_index_inout %Memory[%wo_addr_0] : !hw.inout<uarray<2xi5>>, i1
        sv.passign %4, %wo_data_0 : i5
      }
    }
    hw.output %2 : i5
  }

darthscsi, 2021-09-08:
This is a bug in disallowLocalVariables: https://github.com/llvm/circt/commit/05c7a1310c247cfc00f2607434a63ead47462ae7

darthscsi, 2021-09-08:
PrepareForEmission is generating this to avoid a temporary, which incorrectly introduces a temporary!
```  %Memory = sv.reg  : !hw.inout<uarray<2xi5>>
  %0 = sv.array_index_inout %Memory[%ro_addr_0] : !hw.inout<uarray<2xi5>>, i1
  %1 = sv.read_inout %0 : !hw.inout<i5>
  %x_i5 = sv.constantX : i5
  %2 = comb.mux %ro_en_0, %1, %x_i5 : i5
  %3 = comb.and %wo_en_0, %wo_mask_0 : i1
  %4 = sv.array_index_inout %Memory[%wo_addr_0] : !hw.inout<uarray<2xi5>>, i1
  sv.alwaysff(posedge %wo_clock_0)  {
    sv.if %3  {
      sv.passign %4, %wo_data_0 : i5
    }
  }
  hw.output %2 : i5

lattner, 2021-09-09:
Agreed, I'll take a look.

darthscsi, 2021-09-09:
I have something, I'm just deciding if it's the right thing to do.  (add !isExpressionAlwaysInline at 453)

lattner, 2021-09-09:
oh sorry, I didn't see your comment.  Yep, I landed just that in d8700740.

lattner, 2021-09-09:
ReadInOut is a special case though, because without hoisting that, all the dependent expressions aren't hoistable.

closed 2021-09-09 by lattner