llvm/circt #1712: [Calyx] Verify certains ports are written or driven together.

cgyurgyik, 2021-09-03:
A few examples:
- [ ] A MemoryOp should only be read/written to if its address ports are driven.
- [ ] A RegisterOp should only be written to if the `write_en` port is driven.
- [ ] A binary operation should only be read if both the `lhs` and `rhs` have been driven (similar for unary operation). For this one, it could *technically* be driven in another group, but that would require us to walk the ancestor groups in the control each time we verify. A simpler, more restrictive verification is just ensure that `left` and `right` are driven within the same group.

Calyx does this by means of an attribute in [Calyx 623](https://github.com/cucapra/calyx/pull/623). This could be done by means of attribute, but I imagine just having this as an addition to `verifyGroupOp` is fine as well.

mortbopet, 2021-09-03:
Coincidentally, I should already have the infrastructure for doing this locally (a rewrite of `isPortDrivenByGroup` based on the cell interface).

cgyurgyik, 2021-09-10:
Going to knock this out. May make some slight changes to `isPortDrivenByGroup`, FYI.

PR #1764 (2021-09-14): [Calyx] Verify CombGroupOp is combinational and certain primitive ports are driven together.
This PR focuses on (2) objectives:
1. Verifying that combinational groups are indeed combinational. Combinational components are not supported yet, so component instances are always assumed to be combinational for now. 
2. We also verify that certain ports are written or read together. For example, it does not make sense to write to a register without `write_en` being high. To aid in this, I added helper functions to access certain ports on primitive operations.

While combinational operations may be d [... cut]
closed 2021-09-14 by cgyurgyik