llvm/circt #1715: [HW] Move off of mlir::ModuleOp to hw::DesignOp (or something)

lattner, 2021-09-03:
MLIR upstream is getting progressively better support for the top level op being something other than `mlir::ModuleOp`.  Given that I think we should introduce a new `hw.design` op (better suggestions welcome) to replace this.

This is mostly an idealistic cleanup, but I think it makes sense for a couple of reasons:

1) It would eliminate the confusing and annoying `ModuleOp` vs `HWModuleOp` divide.
2) It would reduce dependencies on the "mlir/IR/BuiltinOps.h" from various places.
3) It gets CIRCT into a more controlled model, where hw::ModuleOp is tied to hw::DesignOp

-Chris

teqdruid, 2021-09-15:
> 1\. It would eliminate the confusing and annoying `ModuleOp` vs `HWModuleOp` divide.

To what are you referring? The namespace issue which is the reason for `hw::HWModuleOp` rather than `hw::ModuleOp`?

> 3\. It gets CIRCT into a more controlled model, where hw::ModuleOp is tied to hw::DesignOp

I don't see a whole lot of practical value in this. Please elaborate.

---

It should be pointed out that this change would be quite pervasive -- it would break every line which finds the top level with `getParentOfType<ModuleOp>()`. Some of this might break silently in cases where you have a `builtin.module` which contains a `hw.design`, which I'd image would be necessary to combine non-CIRCT dialects (e.g. for HLS lowering). It _could_ introduce more complexity in non-CIRCT case due to having to reason about multiple symbol tables (the lookup traversal of which has never made sense to me).  It would also touch most of the tests. (So if we end up doing this, the integration tests should be run before merging to main.) 

While I do agree with the "idealistic cleanup" bit, I still need convincing that this is worth it.

lattner, 2021-09-15:
There is a move in general MLIR to de-builtin-ize things.   The major problem I see here is that the name "module" means something specific in the HW world, and mlir::Module is not it.