llvm/circt #1722: [Calyx] Verify top-level `main` program only has the four required ports.

cgyurgyik, 2021-09-04:
```
calyx.component @main(%go: i1, %clk: i1, %reset: i1) -> (%done: i1) { } // Good
calyx.component @main(%go: i1, %clk: i1, %reset: i1) -> (%done: i1, %bad : i1) { } // Bad
```
For now, I think maintaining this more restrictive requirement on the top-level program is good.

mortbopet, 2021-09-04:
In the Calyx view of the world, this makes good sense. However, I'm worried that this will restrict how Calyx fits into the big-picture MLIR world; while it's probably a long time off, I could imagine that we may have source programs where only certain functions are lowered into Calyx accelerators - other parts of the program may still refer to the SSA values for the in- and output values (ports, streams?) of the (now accelerator-like) function. Obviously there's heaps of other questions regarding this, i.e., how the external world invokes the Calyx accelerator itself, without it being in Calyx dialect, but those are just my initial concerns.

rachitnigam, 2021-09-06:
Similar to @mortbopet, I think it makes sense to require main to have at least those ports but not sure if we need to limit the interface to only those ports.

like any other component, the caller can leave the ports undriven and choose to deal with the consequences as the interface may dictate.

cgyurgyik, 2021-09-10:
Sounds good. Closing.

closed 2021-09-10 by cgyurgyik