llvm/circt #1723: [Calyx] Control program should start with control operator

rachitnigam, 2021-09-05:
The Calyx native parser doesn't accept programs like this:
```
control { Group1; Group2; }
```
And instead requires:
```
control { seq { Group1; Group2 } }
```

This is similar to the problem discovered in #1662 where `while` and `if` cannot have sequences of groups, instead requiring `seq`. I'm starting to think that this requirement is getting onerous so we'll try to remove it on the native compiler end. In the meantime, we should update the emitter to generate `seq { }` around the control program. The `emit.mlir` test for the Calyx dialect is currently not accepted by the native compiler because of this.

cgyurgyik, 2021-09-05:
Hmm the verifier should catch this. I'll look into it.

cgyurgyik, 2021-09-05:
https://github.com/llvm/circt/blob/00a1c209d09aad809d8c4be50d39e55b704e63dd/test/Dialect/Calyx/errors.mlir#L100-L117

Is this what you're referring to?

cgyurgyik, 2021-09-05:
Relevant code:

https://github.com/llvm/circt/blob/00a1c209d09aad809d8c4be50d39e55b704e63dd/lib/Dialect/Calyx/CalyxOps.cpp#L79-L103

cgyurgyik, 2021-09-05:
Closing for now. Please re-open if I'm missing something obvious.

rachitnigam, 2021-09-06:
I'm talking about this test file where the `control` region contains a `par` and a `seq` without a surrounding `seq`: https://github.com/llvm/circt/blob/00a1c209d09aad809d8c4be50d39e55b704e63dd/test/Dialect/Calyx/emit.mlir#L85

You can reproduce the error if you export this file in Calyx format and try to send it to the Calyx Native compiler

PR #1729 (2021-09-06): [Calyx] Verify that multiple control regions are nested in a sequential or parallel operation.
If there are multiple control flow operations in a single control region, they should be nested inside a `calyx.seq` or `calyx.par`. This closes #1723.
commit bec4c85f (2021-09-06): [Calyx] Verify that multiple control regions are nested in a sequential or parallel operation. (#1729)

If there are multiple control flow operations in a single control region, they should be nested 
inside a calyx.seq or calyx.par. This closes #1723.
closed 2021-09-06 by cgyurgyik