llvm/circt #1728: [Calyx] `hw.constant` defined within `calyx.wires`

rachitnigam, 2021-09-06:
While making the Calyx native -> CIRCT flow working, I realized that `hw.constant`, unlike `calyx.instance` etc. are defined within the `calyx.wires` section:
```mlir
  calyx.component @B(%in: i1, %go: i1, %clk: i1, %reset: i1) -> (%out: i1, %done: i1) {
    calyx.wires { 
      %c1 = hw.constant 1 : i1
      calyx.group @Group1 {
        calyx.group_done %c1 : i1
      }
    }
    calyx.control {
      calyx.enable @Group1
    }
  }
```

If I change this to:
```
  calyx.component @B(%in: i1, %go: i1, %clk: i1, %reset: i1) -> (%out: i1, %done: i1) {
    %c1 = hw.constant 1 : i1
    calyx.wires { 
      calyx.group @Group1 {
        calyx.group_done %c1 : i1
      }
    }
    calyx.control {
      calyx.enable @Group1
    }
  }
```
I get the error:
```
emit.mlir:41:24: error: use of undeclared SSA value name
        calyx.group_go %c0 : i1
```

Is there a good reason for this requirement? If not, can we make `hw.constant` be define above the `calyx.wires` section for now?

cgyurgyik, 2021-09-06:
Can you post the full code? From your error, it looks like the constant `%c0` is undeclared. It should be fine to define outside the calyx.wires region now that we've removed calyx.cells

rachitnigam, 2021-09-06:
Ah, my bad. Looks like I was wrong about this.

closed 2021-09-06 by rachitnigam