llvm/circt #1745: [Calyx] Warnings in CalyxOps.cpp

teqdruid, 2021-09-08:
Most of these appear as a result of a function defined in a macro.

clang-11:
```
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:290:2: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
};
 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:298:2: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
};
 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:303:2: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
};
 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:308:2: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
};
 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:938:35: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplUnaryOpCellInterface(PadLibOp);
                                  ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:939:37: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplUnaryOpCellInterface(SliceLibOp);
                                    ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:940:35: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplUnaryOpCellInterface(NotLibOp);
                                  ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:942:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(LtLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:943:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(GtLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:944:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(EqLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:945:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(NeqLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:946:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(GeLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:947:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(LeLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:948:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SltLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:949:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SgtLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:950:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SeqLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:951:34: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SneqLibOp);
                                 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:952:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SgeLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:953:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SleLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:955:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(AddLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:956:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SubLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:957:34: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(ShruLibOp);
                                 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:958:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(RshLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:959:34: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(SrshLibOp);
                                 ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:960:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(LshLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:961:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(AndLibOp);
                                ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:962:32: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(OrLibOp);
                               ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:963:33: warning: extra ';' outside of a function is incompatible with C++98 [-Wc++98-compat-extra-semi]
ImplBinOpCellInterface(XorLibOp);
```

teqdruid, 2021-09-08:
Between https://github.com/llvm/circt/commit/bb5e79b04166a043a037edff2da5366fff84f4e4 and https://github.com/llvm/circt/commit/7893a67e3da6260f9454d05d304a2fa4e6af4fdc these warnings have been cleaned up. Thanks @mortbopet !

closed 2021-09-08 by teqdruid

mortbopet, 2021-09-08:
- and thank you for catching them!