llvm/circt #1749: [Calyx] Warnings in CalyxInterfaces.h.inc

teqdruid, 2021-09-08:
Such a simple construct wasn't added until C++20. We use C++17. (And the rest of MLIR uses 14.)

```
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/CompileControl.cpp:13:
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/PassDetails.h:19:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[16/62] Building CXX object tools/circt/lib/Dialect/Calyx/Transforms/CMakeFiles/obj.CIRCTCalyxTransforms.dir/GoInsertion.cpp.o
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/GoInsertion.cpp:13:
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/PassDetails.h:19:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[20/62] Building CXX object tools/circt/lib/Conversion/CalyxToHW/CMakeFiles/obj.CIRCTCalyxToHW.dir/CalyxToHW.cpp.o
In file included from /home/jodemme/circt/lib/Conversion/CalyxToHW/CalyxToHW.cpp:15:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[21/62] Building CXX object tools/circt/lib/Dialect/Calyx/Export/CMakeFiles/obj.CIRCTExportCalyx.dir/CalyxEmitter.cpp.o
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Export/CalyxEmitter.cpp:15:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[22/62] Building CXX object tools/circt/lib/Dialect/Calyx/CMakeFiles/obj.CIRCTCalyx.dir/CalyxDialect.cpp.o
In file included from /home/jodemme/circt/lib/Dialect/Calyx/CalyxDialect.cpp:14:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[23/62] Building CXX object tools/circt/lib/Dialect/Calyx/Transforms/CMakeFiles/obj.CIRCTCalyxTransforms.dir/RemoveGroups.cpp.o
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/RemoveGroups.cpp:13:
In file included from /home/jodemme/circt/lib/Dialect/Calyx/Transforms/PassDetails.h:19:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
2 warnings generated.
[27/62] Building CXX object tools/circt/lib/Dialect/Calyx/CMakeFiles/obj.CIRCTCalyx.dir/CalyxOps.cpp.o
In file included from /home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:13:
In file included from /home/jodemme/circt/include/circt/Dialect/Calyx/CalyxOps.h:116:
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:117:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((*static_cast<ConcreteOp *>(this))->getContext(), name),
              ^
tools/circt/include/circt/Dialect/Calyx/CalyxInterfaces.h.inc:274:15: warning: designated initializers are a C++20 extension [-Wc++20-designator]
              .name=StringAttr::get((llvm::cast<ConcreteOp>(tablegen_opaque_val))->getContext(), name),
              ^
/home/jodemme/circt/lib/Dialect/Calyx/CalyxOps.cpp:288:18: warning: designated initializers are a C++20 extension [-Wc++20-designator]
        PortInfo{.name = portNamesAttr[i].cast<StringAttr>(),
                 ^
```

mikeurbach, 2021-09-08:
@cgyurgyik FYI^. I'm in the middle of recompiling LLVM but once my laptop cools off I can also take this.

cgyurgyik, 2021-09-08:
Mike did indeed [call me out on this](#include "contentads/analysis/hero/shared/scaffolding/handlers/base_http_handler.h") :). 

Naive question; is this just generally supported by the clang compiler? I've used this in < C++17 (and I think 14 as well).

>  but once my laptop cools off I can also take this.

Thanks! I'll be back on my non-work laptop later tonight if you're suddenly occupied.

youngar, 2021-09-08:
Look like it got handled: https://github.com/llvm/circt/commit/5fbf9974e1543ab50dc5c89a6fc6ea6b7ed0c5b4

youngar, 2021-09-08:
Windows build is green again and I don't see any warnings in the Linux build either, so I'm going to close this.

closed 2021-09-08 by youngar

cgyurgyik, 2021-09-08:
Thank you @seldridge !

seldridge, 2021-09-08:
No problem! I'm still not sure why clang or gcc aren't erroring like msvcc here. 🤔