llvm/circt #1758: [FIRRTL] Preserve names until annotation lowering is done

darthscsi, 2021-09-09:
Some annotations produce binds and XMRs which are much nicer with names.  We don't know if there are annotations which will prefer a name until after parsing when we've done a first pass scattering annotations (some of which may add don't touch to existing ops).

There is a multipart set of changes for this:
Part 1: move name-dropping into a a canonicalization from the parser.
Part 2: Unify annotation scattering/early processing into a post-parse, pre-canonicalize pass which ensures that IR is sufficiently annotated/mutated so that further passes (including dropping names) won't affect annotation-dependent passes.

seldridge, 2021-09-09:
:+1: This would fix my concerns from https://github.com/llvm/circt/pull/1558#issuecomment-908893039.

darthscsi, 2021-09-10:
> 👍 This would fix my concerns from [#1558 (comment)](https://github.com/llvm/circt/pull/1558#issuecomment-908893039).

It is just pulling out code from that PR and committing it separately.  But @youngar already hit cases where early node elimination caused semantic differences.