llvm/circt #1769: [Calyx] Inlining `%wire` incorrectly assigns guard statements to ports

rachitnigam, 2021-09-10:
The following MLIR code:
```mlir
module  {
  calyx.program  {
    calyx.component @main(%in0: i32, %in1: i32, %in2: i32, %clk: i1, %reset: i1, %go: i1) -> (%out0: i32, %done: i1) {
      %true = hw.constant true
      %c0_i32 = hw.constant 0 : i32
      %while_0_arg2_reg.in, %while_0_arg2_reg.write_en, %while_0_arg2_reg.clk, %while_0_arg2_reg.reset, %while_0_arg2_reg.out, %while_0_arg2_reg.done = calyx.register "while_0_arg2_reg" : i32, i1, i1, i1, i32, i1
      %while_0_arg1_reg.in, %while_0_arg1_reg.write_en, %while_0_arg1_reg.clk, %while_0_arg1_reg.reset, %while_0_arg1_reg.out, %while_0_arg1_reg.done = calyx.register "while_0_arg1_reg" : i32, i1, i1, i1, i32, i1
      %while_0_arg0_reg.in, %while_0_arg0_reg.write_en, %while_0_arg0_reg.clk, %while_0_arg0_reg.reset, %while_0_arg0_reg.out, %while_0_arg0_reg.done = calyx.register "while_0_arg0_reg" : i32, i1, i1, i1, i32, i1
      %bb2_arg1_reg.in, %bb2_arg1_reg.write_en, %bb2_arg1_reg.clk, %bb2_arg1_reg.reset, %bb2_arg1_reg.out, %bb2_arg1_reg.done = calyx.register "bb2_arg1_reg" : i32, i1, i1, i1, i32, i1
      %bb2_arg0_reg.in, %bb2_arg0_reg.write_en, %bb2_arg0_reg.clk, %bb2_arg0_reg.reset, %bb2_arg0_reg.out, %bb2_arg0_reg.done = calyx.register "bb2_arg0_reg" : i32, i1, i1, i1, i32, i1
      calyx.wires  {
        calyx.group @assign_while_0_init  {
          calyx.assign %while_0_arg0_reg.in = %in0 : i32
          calyx.assign %while_0_arg0_reg.write_en = %true : i1
          calyx.assign %while_0_arg1_reg.in = %c0_i32 : i32
          calyx.assign %while_0_arg1_reg.write_en = %true : i1
          calyx.assign %while_0_arg2_reg.in = %c0_i32 : i32
          calyx.assign %while_0_arg2_reg.write_en = %true : i1
          %0 = comb.and %while_0_arg0_reg.done, %while_0_arg1_reg.done : i1
          %1 = comb.and %0, %while_0_arg2_reg.done : i1
          calyx.group_done %1 : i1
        }
      }
      calyx.control  {
        calyx.enable @assign_while_0_init
      }
    }
  }
}
```

Generates the following assignment in the Calyx code using `circt-translate --export-calyx`:
```
      assign_while_0_init[done] = ((while_0_arg0_reg.done & while_0_arg1_reg.done) & while_0_arg2_reg.done);
```
This is incorrect. Calyx does not allow for complex guard statements to be assigned to ports. It should instead be:
```
 assign_while_0_init[done] = ((while_0_arg0_reg.done & while_0_arg1_reg.done) & while_0_arg2_reg.done) ? 1'd1;
```

The problem is that that guard inliner for the emitter incorrectly transforms `calyx.group_done %1 : i1` into the above by recursively inlining `%1`.

Two possible solutions are:
1. Always using `calyx.group_done %true, %1 ?` or something so that there is an explicit port
2. Making the emitter less smart and just structurally instantiate the wires into the relevant subcomponent like `std_add` etc.

mikeurbach, 2021-09-11:
This is something I've found a little suprising myself.

Is this

```
group_name[done] = <complex expr> ? 1'd1
```

Semanticaly different than

```
group_name[done] = <complex expr>
```

I'm trying to understand why adding a constant guard makes the complex expression legal. Since the gaurd is a constant, shouldn't they be equivalent, and the un-guarded version legal? I'm sure I'm missing something...

rachitnigam, 2021-09-11:
Unfortunately not. If you have an unguarded assignment, the native compiler expects a port as the source.

The whole business with named wires in the CIRCT dialect is because we eventually want to homogenize the representation and just allow for ports in both the guard and the source positions.

mortbopet, 2021-09-11:
> Always using calyx.group_done %true, %1 ? or something so that there is an explicit port

This should be an easy change in the lowering pass. However, i agree that it is a bit smelly that we're able to generate code which is a valid Calyx dialect program (as per operation validators), but exports to something which is not a valid native program.

cgyurgyik, 2021-09-11:
IMO as @mortbopet just pointed out, this should be a verification. The same thing applies to the `go` hole.

mikeurbach, 2021-09-11:
> the native compiler expects a port as the source

I see. And if the assignment is guarded, a more complex source is allowed? I'm wondering why that is the case, and whether its worth revisiting this restriction. I haven't looked into the details, but it sounds like the native compiler does something differently when a guard is present (even if it's just a constant). Which makes me think it could be possible to do that thing regardless of whether the assignment is guarded. Again, I don't have all the background here, but coming from a fresh perspective, it seems a little weird to insert a constant guard that's always true in order to create valid IR.

Anyway, for now, it's simple enough to just update the verifier on the CIRCT side and generate guards when needed. Just some thoughts for discussion.

cgyurgyik, 2021-09-11:
> I see. And if the assignment is guarded, a more complex source is allowed?

If I remember correctly, if the assignment is guarded, you still cannot have a complex source.

In CIRCT, a complex expression just means `hw::{xor, and, or}`.
```
group_name[done] = guard ? <complex expr>; // Not allowed
some_port = guard ? <complex expr>;        // Not allowed
```

I think this boils down to complex expressions cannot be the source of an assignment or group port.

cgyurgyik, 2021-09-11:
To better illustrate:
```mlir
import "primitives/std.lib";

component main() -> () {
  cells {
    a = std_reg(2);
  }
  wires {
    group A {
      a.in = 2'd1;
      a.write_en = 1'b1;
      A[done] = (a.done & 1'b1) ? 1'b1;           // <-- Allowed
      // A[done] =  1'b1;                            <-- Not allowed, because then this is just a combinational group.
      // A[done] = 1'b1 ? (a.done & 1'b1);           <-- Not allowed, complex source.
      // A[done] = (a.done & 1'b1) ? (1'b1 & 1'b1);  <-- Not allowed, complex source.
      // A[done] = (a.done & 1'b1);                  <-- Not allowed, complex source.
    }
  }
  control { A; }
}
```

cgyurgyik, 2021-09-11:
Added verifiers in #1774.

rachitnigam, 2021-09-12:
Chris’s explanation is correct. I think in the next few weeks, we’re going to try to add wire declarations in Calyx native and make it so that both guards and source ports only accept simple expressions.

PR #1774 (2021-09-14): [Calyx] Verify that sources of group ports and assignments are not logically complex.
As discussed in #1769, some verifiers should be added to avoid giving the sources of group ports / assignments logically complex expressions. Also adds a verifier to ensure that `GroupOp` isn't being used as a `CombGroupOp`, and makes the necessary refactoring.

```mlir
import "primitives/std.lib";

component main() -> () {
  cells {
    a = std_reg(2);
  }
  wires {
    group A {
      a.in = 2'd1;
      a.write_en = 1'b1;
      A[done] = (a.done & 1'b1) ? 1'b1;           // <-- 1. Allowed
      // A[done] = [... cut]
closed 2021-09-14 by cgyurgyik