llvm/circt #1775: [ESI] Make instances match module declarations

lattner, 2021-09-13:
We have a problem where `InstanceOp::verifySymbolUses` is unable to enforce the instances and module declarations line up w.r.t. port names and types.  This is because ESI is (ab)using the lack of checking by using one external declaration and having multiple instances refer to it.  A consequence of this is that [this patch](https://github.com/llvm/circt/commit/8a41c3f326436427a1dc6da792c6cad475c11826) had to land an important patch on the path to parameterized modules.

Please fix ESI so it creates one extern module for each signature it needs.  These can/should have the same "verilogName" attribute so they get emitted as the same declaration, but until the code is structured and verified, I can't make progress with parameterized modules.

Thanks!

teqdruid, 2021-09-13:
To be fair, this is a new requirement. When this exact issue has come up in the past (IIRC) it was discussed and this was determined to be the way to do it. There's a reason the `InstanceOp` verifier doesn't check types on extern mods.

I'll get on it.

lattner, 2021-09-14:
I recall we discussed something here, but I don’t recall the details.  Thank you for taking care of this.  The lack of checking is hiding other problems.

teqdruid, 2021-09-14:
I'm a big fan of static verification, but just figured it'd have to be put off until we had support for parameterized extern modules. Didn't even think of creating one extern mod for each parameterization! This is a far nicer design.

lattner, 2021-09-14:
> This is a far nicer design.

I wouldn't say that it is "nice", but it is "consistent" with our current parameterization approach.  Thank you for fixing this!

PR #1780 (2021-09-14): [ESI] Make instances match module declarations
Closes #1775 .

Also allows us to have the `hw::InstanceOp` verifier not ignore `hw::HWModuleExternOp`s.
closed 2021-09-14 by lattner