llvm/circt #1776: [FIRRTL] `FIRRTLLowering::visitDecl(MemOp op)`  has no tests

lattner, 2021-09-13:
This code is doing a somewhat fiddly thing with memory modules - creating instances before the referenced module is created.  I need to refactor this code to make progress on parameterized module but the logic has no tests.  Can someone add a test or two so I don't break anything obvious?

lattner, 2021-09-13:
@darthscsi can you assign this to someone please?  Thanks

lattner, 2021-09-13:
Specific issue - if you add an abort to the end of this method, no tests fail.

lattner, 2021-09-13:
@

darthscsi, 2021-09-13:
hmm, lower-to-hw.mlir:657 should be testing this.  Will look at it.

darthscsi, 2021-09-13:
When I add an abort to the end of visitDecl, I see it in the test case:
```andrew@narya:~/src/circt/build$ /home/andrew/src/circt/build/bin/circt-opt -lower-firrtl-to-hw /home/andrew/src/circt/test/Conversion/FIRRTLToHW/lower-to-hw.mlir
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Aborted```

I actually see 4 tests fail:
```  CIRCT :: Conversion/FIRRTLToHW/errors.mlir
  CIRCT :: Conversion/FIRRTLToHW/lower-to-hw-errors.mlir
  CIRCT :: Conversion/FIRRTLToHW/lower-to-hw.mlir
  CIRCT :: Dialect/FIRRTL/SFCTests/data-taps.fir```

darthscsi, 2021-09-13:
Without the abort, tests pass:
```
andrew@narya:~/src/circt/build$ ninja check-circt
[165/166] Running the CIRCT regression tests

Testing Time: 1.49s
  Unsupported      :   2
  Passed           : 258
  Expectedly Failed:   3
  ```

closed 2021-09-13 by darthscsi

lattner, 2021-09-13:
You're totally right, sorry for the noise!