llvm/circt #1777: [Calyx] [Tracker] Make SCF to Calyx work with native compiler

rachitnigam, 2021-09-13:
Tracker issue to make code generated by #1630 work with the native compiler. Let's consider this successfully done when we can simulate SCF generated program on Verilator by going through the Calyx native compiler

## CIRCT
- [ ] #1769

## Calyx

mortbopet, 2021-09-23:
@cgyurgyik @rachitnigam any final comments on this?

cgyurgyik, 2021-09-23:
I imagine that we'll run into bugs while lowering to Calyx, but I'm fine opening separate issues for this as well.

rachitnigam, 2021-09-23:
Yeah, I agree with Chris. Let's ensure that at least the tests in the CIRCT repo compile with the native compiler correctly first and then we can try moving onto bigger examples.

rachitnigam, 2021-12-21:
Looks like this is done for now so closing it!

closed 2021-12-21 by rachitnigam