llvm/circt #1788: [Python] Update InstanceBuilder to populate input/output names for hw.InstanceOp

mikeurbach, 2021-09-14:
After https://github.com/llvm/circt/commit/3d19aa785468d4819961acf296c015d6f7fb318f, `HWInstanceOp` (in C++) has two new attributes to store arrays of input/output port names. Since this changed in ODS, the generated `hw.InstanceOp` (in Python) has two new arguments to the constructor. We need to update this to pass them in accordingly as `post_args`:

https://github.com/llvm/circt/blob/8b08c41e6cc5b6765b34ba1d037667881d69191a/lib/Bindings/Python/circt/dialects/_hw_ops_ext.py#L34

This manifests as errors in the Python bindings tests that look like this:

```
TypeError: __init__() missing 2 required positional arguments: 'parameters' and 'sym_name'
```

mikeurbach, 2021-09-14:
Filing this since I noticed it. I won't be able to take a look immediately, so if anyone else jumps on it let me know. Otherwise, I can take a look this week since I've owned the Python bindings for building InstanceOps so far.

teqdruid, 2021-09-15:
Yeah... this is one of those times when it'd be _really_ nice to be able to use C++ builders...

teqdruid, 2021-09-15:
I'm on it.

closed 2021-09-15 by teqdruid

lattner, 2021-09-15:
Thx John!  I'd also recommend following the model where we prefer people to construct an instance with a module, and therefore pull forward the symbol and names automatically.  It is more convenient and more future proof. 

The low level API is still useful for specific cases of course.

mikeurbach, 2021-09-15:
>  construct an instance with a module

Yep, that's more or less what this InstanceBuilder thing does, and why the update here was straightforward.

commit cd8d55a6 (2021-09-15): [Python] Fixing `InstanceOp` binding and ditching module decorator

Fixes the integration tests. Closes #1788.