llvm/circt #1799: [FIRRTL] Grand Central Views Needs to Mangle Verilog Keywords in XMRs

seldridge, 2021-09-15:
Currently, if a field in a Grand Central-generated interface conflicts with a Verilog keyword, this will get mangled properly during `HWLegalizeNames`. However, this will not update the XMR that has already been dumped into an `sv.verbatim` op.

The simple solution would be to mangle names at the point at which they are created so that the verbatim XMRs will be correct.

Concretely, I'm seeing this issue for an interface field called `signed` where it is mangled to `signed_0`, but the XMR still points to `foo.bar.signed`.

lattner, 2021-09-16:
I think @prithayan 's work on verbatim will allow you to put a symbol into the verbatim string so it tracks name updates etc?

seldridge, 2022-07-12:
This should be fixed now that we use `sv::VerbatimOp` with symbols to generate the XMRs.

closed 2022-07-12 by seldridge