llvm/circt #1800: HLS scheduling and pipeline creation tasks

mikeurbach, 2021-09-16:
This is a tracking issue to aggregate some of the work in progress to set up an HLS frontend into CIRCT. The scope covers scheduling loops and creating a pipeline representation, which should be further lowered into Calyx. This has been discussed in a few places, but I wanted to have a single location to track the tasks for visibility. Links to smaller issues/PRs will be added as they come. I'm planning to tackle these, but I'm all ears for any feedback or collaboration.

- [ ] Affine dependency analysis pass to compute memory access dependencies

* In order to inform scheduling, we need to get memory access dependencies. This can be done using `checkMemRefAccessDependence`, as shown [here](https://llvm.discourse.group/t/rfc-representing-pipelined-loops/4171/13?u=mikeurbach).
* This is necessarily restricted to Affine loops. I think this covers all the cases a frontend generating Linalg will produce, as well as cases supported by Polygeist's support for raising SCF loops to Affine loops. For other more exotic situations, I don't think there is any dependence analysis supported by MLIR out of the box. Perhaps we could work towards something like LLVM's [dependence analysis](https://llvm.org/docs/Passes.html#da-dependence-analysis) to support these cases.
* The results of this analysis can either be persisted in the IR as attributes, or as an Analysis that is preserved for later passes. I'm not sure if there are other options, or which is preferable.
* This is not necessarily CIRCT specific, but it may be prudent to incubate this here and see if the community is interested in having such a pass upstream.

- [ ] Component library to inform scheduling decisions

* See [this](https://llvm.discourse.group/t/infrastructure-for-converting-standard-to-comb-or-primitives/4224) for more discussion.
* In order to support scheduling (and eventually lowering to hardware primitives), it would be nice to have a "component library".
* Such a component library should answer a few important questions:
  * Can a given operation be lowered into a hardware primitive?
  * What hardware primitive does a given operation lower into?
  * What is the latency of said primitive?
  * What is its II, if any?
  * What are its resource limits?
* I have a prototype component library to try out some ideas, and I have enough experimentation to open an issue for further discussion.

- [ ] Scheduling analysis pass

* Given the above, we should have what we need to create and solve a scheduling problem at the level of Affine loops
* This pass will also need to persist its results in some way, either in the IR or as an Analysis.

- [ ] StaticLogic pipeline while loop operation

* See [this](https://llvm.discourse.group/t/rfc-representing-pipelined-loops/4171) for more discussion.
* PR here: #1759 

- [ ] StaticLogic pipeline while loop lowering pass

* Given the above analyses and operation, we have all the information we need to create a pipeline loop.
* This could start directly from Affine loops, as this is the only place we can easily compute memory access dependencies. Or, we could lower from Affine to SCF (and convert for loops to while loops). For full generality, a converting an SCF while loop to a StaticLogic pipeline while loop seems best, but any memory accesses will have to have dependence information analyzed at the Affine level, at least for now.
* Pragmatically, it may be easier to write tests for this pass based on attributes attached to Affine/SCF IR, which could be one reason to persist Affine/Scheduling analyses as attributes in the IR

cgyurgyik, 2021-09-16:
Throwing the Calyx tag on here as well; it seems pertinent after reading:

> The scope covers scheduling loops and creating a pipeline representation, which should be further lowered into Calyx.

mikeurbach, 2021-11-12:
Sorry for the radio silence on this, but I've just updated the checklist and filed a PR for the first version of a pipeline lowering pass: https://github.com/llvm/circt/pull/2157

mikeurbach, 2022-02-08:
After https://github.com/llvm/circt/commit/40108faff3a370c4f63ed7639b5f88e5a5eaebaf, I think we can consider the CIRCT support necessary for a first end-to-end flow complete.

The flow supports single Affine loops, including iter_args and returns, like a dot product: https://github.com/llvm/circt/blob/40108faff3a370c4f63ed7639b5f88e5a5eaebaf/test/Conversion/AffineToStaticLogic/loops.mlir#L56-L63.

These are succesfully scheduled for the Calyx operator library, lowered through StaticLogic to Calyx, lowered in the native Calyx compiler, and exported to Verilog. I've verified the functional correctness of the dot example with the Calyx interpreter as well as Verilator. Unfortunately, the hardware does not show overlapping execution of pipeline stages in the cycle-by-cycle simulation, but this is a known missing feature in Calyx that I'm helping add back: https://github.com/cucapra/calyx/issues/654.

closed 2022-02-08 by mikeurbach