llvm/circt #1802: [Calyx] Missing emission of 'else' branches

mortbopet, 2021-09-16:
In the following program, the "else" branch of the control `if`-statement missing in the emitted native Calyx IR, when running `circt-translate --export-calyx`:
```mlir
module  {
  calyx.program  {
    calyx.component @main(%in0: i32, %in1: i32, %clk: i1 {clk}, %reset: i1 {reset}, %go: i1 {go}) -> (%out0: i32, %done: i1 {done}) {
      %true = hw.constant true
      %std_add_2.left, %std_add_2.right, %std_add_2.out = calyx.std_add "std_add_2" : i32, i32, i32
      %std_add_1.left, %std_add_1.right, %std_add_1.out = calyx.std_add "std_add_1" : i32, i32, i32
      %std_add_0.left, %std_add_0.right, %std_add_0.out = calyx.std_add "std_add_0" : i32, i32, i32
      %std_sub_1.left, %std_sub_1.right, %std_sub_1.out = calyx.std_sub "std_sub_1" : i32, i32, i32
      %std_sub_0.left, %std_sub_0.right, %std_sub_0.out = calyx.std_sub "std_sub_0" : i32, i32, i32
      %std_ge_0.left, %std_ge_0.right, %std_ge_0.out = calyx.std_ge "std_ge_0" : i32, i32, i1
      %ret_arg0_reg.in, %ret_arg0_reg.write_en, %ret_arg0_reg.clk, %ret_arg0_reg.reset, %ret_arg0_reg.out, %ret_arg0_reg.done = calyx.register "ret_arg0_reg" : i32, i1, i1, i1, i32, i1
      %bb3_arg1_reg.in, %bb3_arg1_reg.write_en, %bb3_arg1_reg.clk, %bb3_arg1_reg.reset, %bb3_arg1_reg.out, %bb3_arg1_reg.done = calyx.register "bb3_arg1_reg" : i32, i1, i1, i1, i32, i1
      %bb3_arg0_reg.in, %bb3_arg0_reg.write_en, %bb3_arg0_reg.clk, %bb3_arg0_reg.reset, %bb3_arg0_reg.out, %bb3_arg0_reg.done = calyx.register "bb3_arg0_reg" : i32, i1, i1, i1, i32, i1
      %bb2_arg0_reg.in, %bb2_arg0_reg.write_en, %bb2_arg0_reg.clk, %bb2_arg0_reg.reset, %bb2_arg0_reg.out, %bb2_arg0_reg.done = calyx.register "bb2_arg0_reg" : i32, i1, i1, i1, i32, i1
      %bb1_arg0_reg.in, %bb1_arg0_reg.write_en, %bb1_arg0_reg.clk, %bb1_arg0_reg.reset, %bb1_arg0_reg.out, %bb1_arg0_reg.done = calyx.register "bb1_arg0_reg" : i32, i1, i1, i1, i32, i1
      calyx.wires  {
        calyx.assign %out0 = %ret_arg0_reg.out : i32
        calyx.group @bb0_to_bb1  {
          calyx.assign %bb1_arg0_reg.in = %in0 : i32
          calyx.assign %bb1_arg0_reg.write_en = %true : i1
          calyx.group_done %bb1_arg0_reg.done : i1
        }
        calyx.group @bb0_to_bb2  {
          calyx.assign %bb2_arg0_reg.in = %in1 : i32
          calyx.assign %bb2_arg0_reg.write_en = %true : i1
          calyx.group_done %bb2_arg0_reg.done : i1
        }
        calyx.group @bb1_to_bb3  {
          calyx.assign %bb3_arg0_reg.in = %std_sub_0.out : i32
          calyx.assign %bb3_arg0_reg.write_en = %true : i1
          calyx.assign %bb3_arg1_reg.in = %std_sub_1.out : i32
          calyx.assign %bb3_arg1_reg.write_en = %true : i1
          calyx.assign %std_sub_0.left = %bb1_arg0_reg.out : i32
          calyx.assign %std_sub_0.right = %in1 : i32
          calyx.assign %std_sub_1.left = %bb1_arg0_reg.out : i32
          calyx.assign %std_sub_1.right = %in0 : i32
          %0 = comb.and %bb3_arg0_reg.done, %bb3_arg1_reg.done : i1
          calyx.group_done %true, %0 ? : i1
        }
        calyx.group @bb2_to_bb3  {
          calyx.assign %bb3_arg0_reg.in = %std_add_0.out : i32
          calyx.assign %bb3_arg0_reg.write_en = %true : i1
          calyx.assign %bb3_arg1_reg.in = %std_add_1.out : i32
          calyx.assign %bb3_arg1_reg.write_en = %true : i1
          calyx.assign %std_add_0.left = %bb2_arg0_reg.out : i32
          calyx.assign %std_add_0.right = %in1 : i32
          calyx.assign %std_add_1.left = %bb2_arg0_reg.out : i32
          calyx.assign %std_add_1.right = %in0 : i32
          %0 = comb.and %bb3_arg0_reg.done, %bb3_arg1_reg.done : i1
          calyx.group_done %true, %0 ? : i1
        }
        calyx.group @ret_assign_0  {
          calyx.assign %ret_arg0_reg.in = %std_add_2.out : i32
          calyx.assign %ret_arg0_reg.write_en = %true : i1
          calyx.assign %std_add_2.left = %bb3_arg0_reg.out : i32
          calyx.assign %std_add_2.right = %bb3_arg1_reg.out : i32
          calyx.group_done %ret_arg0_reg.done : i1
        }
        calyx.comb_group @comb_bb0_0  {
          calyx.assign %std_ge_0.left = %in0 : i32
          calyx.assign %std_ge_0.right = %in1 : i32
        }
      }
      calyx.control  {
        calyx.seq  {
          calyx.if %std_ge_0.out with @comb_bb0_0  {
            calyx.seq  {
              calyx.enable @bb0_to_bb1 {compiledGroups = []}
              calyx.enable @bb1_to_bb3 {compiledGroups = []}
            }
          } else  {
            calyx.seq  {
              calyx.enable @bb0_to_bb2 {compiledGroups = []}
              calyx.enable @bb2_to_bb3 {compiledGroups = []}
            }
          }
          calyx.enable @ret_assign_0 {compiledGroups = []}
        }
      }
    }
  }
}
```

emits to the native calyx IR:
```
import "primitives/core.futil";
component main(in0: 32, in1: 32, @clk clk: 1, @reset reset: 1, @go go: 1) -> (out0: 32, @done done: 1) {
  cells {
    std_add_2 = std_add(32);
    std_add_1 = std_add(32);
    std_add_0 = std_add(32);
    std_sub_1 = std_sub(32);
    std_sub_0 = std_sub(32);
    std_ge_0 = std_ge(32);
    ret_arg0_reg = std_reg(32);
    bb3_arg1_reg = std_reg(32);
    bb3_arg0_reg = std_reg(32);
    bb2_arg0_reg = std_reg(32);
    bb1_arg0_reg = std_reg(32);
  }
  wires {
    out0 = ret_arg0_reg.out;
    group bb0_to_bb1 {
      bb1_arg0_reg.in = in0;
      bb1_arg0_reg.write_en = 1'd1;
      bb0_to_bb1[done] = bb1_arg0_reg.done;
    }
    group bb0_to_bb2 {
      bb2_arg0_reg.in = in1;
      bb2_arg0_reg.write_en = 1'd1;
      bb0_to_bb2[done] = bb2_arg0_reg.done;
    }
    group bb1_to_bb3 {
      bb3_arg0_reg.in = std_sub_0.out;
      bb3_arg0_reg.write_en = 1'd1;
      bb3_arg1_reg.in = std_sub_1.out;
      bb3_arg1_reg.write_en = 1'd1;
      std_sub_0.left = bb1_arg0_reg.out;
      std_sub_0.right = in1;
      std_sub_1.left = bb1_arg0_reg.out;
      std_sub_1.right = in0;
      bb1_to_bb3[done] = (bb3_arg0_reg.done & bb3_arg1_reg.done) ? 1'd1;
    }
    group bb2_to_bb3 {
      bb3_arg0_reg.in = std_add_0.out;
      bb3_arg0_reg.write_en = 1'd1;
      bb3_arg1_reg.in = std_add_1.out;
      bb3_arg1_reg.write_en = 1'd1;
      std_add_0.left = bb2_arg0_reg.out;
      std_add_0.right = in1;
      std_add_1.left = bb2_arg0_reg.out;
      std_add_1.right = in0;
      bb2_to_bb3[done] = (bb3_arg0_reg.done & bb3_arg1_reg.done) ? 1'd1;
    }
    group ret_assign_0 {
      ret_arg0_reg.in = std_add_2.out;
      ret_arg0_reg.write_en = 1'd1;
      std_add_2.left = bb3_arg0_reg.out;
      std_add_2.right = bb3_arg1_reg.out;
      ret_assign_0[done] = ret_arg0_reg.done;
    }
    comb group comb_bb0_0 {
      std_ge_0.left = in0;
      std_ge_0.right = in1;
    }
  }
  control {
    seq {
      if std_ge_0.out with comb_bb0_0 {
        seq {
          bb0_to_bb1;
          bb1_to_bb3;
        }
      }
      ret_assign_0;
    }
  }
}
```

cgyurgyik, 2021-09-16:
Just to clarify, the only issue is that the `else` region isn't being emitted. Is this correct?

mortbopet, 2021-09-16:
Yep, the remainder looks correct to me.

cgyurgyik, 2021-09-16:
Sounds good! In the future, it can be helpful to make the failing example as small as possible, and, when useful, provide the correct behavior as well. In any case, this is probably a simple fix of emitting the `else` region, if it exists. Thanks for spotting this.

cgyurgyik, 2021-09-16:
Offending code here:

https://github.com/llvm/circt/blob/072abc36c9a2347c50c8a86efb80f6f27405e8cd/lib/Dialect/Calyx/Export/CalyxEmitter.cpp#L385-L392

`emitCalyxControl` recursively emits, however it doesn't look at the `then` and `else` case of the `IfOp`. Probably cleaner to split up the `IfOp` and `WhileOp` case. Feel free to take it if someone has a chance. If not, I'll get to it when I can.

PR #1804 (2021-09-17): [Calyx] [NFC] Emitter now emits `ElseRegion`.
Closes #1802.
closed 2021-09-17 by cgyurgyik