llvm/circt #1810: [Python] [CI Integration Tests] Bindings broken

teqdruid, 2021-09-19:
https://github.com/llvm/circt/runs/3640159047?check_suite_focus=true

According to `git bisect`,
```
471e60ceafc8dc43fa7e582ca35d5830ed6a0aa9 is the first bad commit
commit 471e60ceafc8dc43fa7e582ca35d5830ed6a0aa9
Author: Chris Lattner <clattner@nondot.org>
Date:   Fri Sep 17 17:48:38 2021 -0700

    [HW] Add support for parameter decls to the hw.module* operations.

    This adds IR support to parse and print them, but there is no export
    verilog support yet.
```

FYI @lattner

youngar, 2021-09-20:
PR here: https://github.com/llvm/circt/pull/1817

teqdruid, 2021-09-20:
Now we're back to the original errors:

```
Failed Tests (8):
  CIRCT :: Bindings/Python/dialects/comb.py
  CIRCT :: Bindings/Python/dialects/esi.py
  CIRCT :: Bindings/Python/dialects/hw.py
  CIRCT :: Bindings/Python/dialects/msft.py
  CIRCT :: Bindings/Python/dialects/rtl.py
  CIRCT :: Bindings/Python/dialects/rtl_errors.py
  CIRCT :: Bindings/Python/dialects/seq.py
  CIRCT :: Bindings/Python/support/connect.py
```

teqdruid, 2021-09-23:
I'm going to take a whack at fixing this.

PR #1858 (2021-09-23): [Python] Fix Python Bindings
Closes #1810 

The changes to `hw.module` and `hw.instance` make them incompatible with `msft.module` and `msft.instance` as far as the python bindings are concerned. I'm not sure what I want to do here, though I'll likely update the msft variants. I've marked that test as `xfail` for the moment.
commit 99f0092c (2021-09-23): [Python] Fix Python Bindings (#1858)

Closes #1810 

The changes to `hw.module` and `hw.instance` make them incompatible with `msft.module` and `msft.instance` as far as the python bindings are concerned. I'm not sure what I want to do here so I've marked that test as `xfail` for the moment.
closed 2021-09-23 by teqdruid