llvm/circt #1828: [FIRRTL] BlackBoxInlineAnno file path

darthscsi, 2021-09-21:
Annotation:
    "class":"firrtl.transforms.BlackBoxInlineAnno",
    "target":"CoreIPSubsystemVerifTestHarness.DummyVroom",
Needs to pay attention to:
    "class":"sifive.enterprise.firrtl.TestBenchDirAnnotation",
to place files in correct directory

seldridge, 2021-09-21:
I may be wrong here, but I think `BlackBoxInlineAnno` will default to `"."` unless a `BlackBoxTargetDirAnno` is present. I'm surprised that `TestBenchDirAnnotation` (which is purely internal) matters.

seldridge, 2021-09-21:
I see what's going on here...

You are exactly correct about `sifive.enterprise.firrtl.TestBenchDirAnnotation`. Our internal test code extraction is locally overwriting the `BlackBoxTargetDirAnno` with the contents of this annotation, but only conditionally doing it.

We will need some special handling to do this, but it would be incorrect to blindly change the behavior of `BlackBoxInlineAnno` based on this other annotation. 😬

darthscsi, 2021-09-21:
I suspected it was target-dependent.

darthscsi, 2021-09-23:
looks like a general difference in output location, SFC is adding directories which firtool isn't.  This probably can be handled in build scripts.

closed 2021-09-23 by darthscsi