llvm/circt #1833: Verilator tests not disabling

lattner, 2021-09-21:
I updated and have two new tests in test/verilator that are failing (main.sv and errors1.sv).  It's weird though, they are correctly marked with `REQUIRES: verilator` line, and cmake seems to realize I don't have verilator:

```
-- Using MLIRConfig.cmake in: /Users/chrisl/Projects/circt/llvm/build/lib/cmake/mlir
-- Using LLVMConfig.cmake in: /Users/chrisl/Projects/circt/llvm/build/lib/cmake/llvm
-- Building with -fPIC
-- Disabling Verilator tests.
-- Did not find vivado.
-- Did not find Quartus.
...
```

However, when running the tests, lit tries to run them even though I don't have a `circt/llvm/build/./bin/verilator` executable:

```
******************** TEST 'CIRCT :: verilator/main.sv' FAILED ********************
Script:
--
: 'RUN: at line 2';   /Users/chrisl/Projects/circt/llvm/build/./bin/verilator --cc --top-module main -Wall -Wpedantic /Users/chrisl/Projects/circt/test/verilator/main.sv
--
Exit Code: 127

Command Output (stderr):
--
/Users/chrisl/Projects/circt/build/test/verilator/Output/main.sv.script: line 1: /Users/chrisl/Projects/circt/llvm/build/./bin/verilator: No such file or directory
```

Am I doing something wrong here or are verilator tests missing something to get lit to disable them correctly?

teqdruid, 2021-09-22:
The only two recent changes to the root CMakeLists.txt or lit configs are: 34d3ed999e6b5a7bddd09d5d5d4ea69e1bd1b61d and 2829895862c89284f5ea6bea8afafcf1dab484c7. I'm guessing it was the former which broke it.

lattner, 2021-09-22:
Indeed, great catch, thanks!

6d96350d

closed 2021-09-22 by lattner

youngar, 2021-09-22:
Thanks for fixing this!